mikakaraila/node-red-contrib-opcua #432: Debug: OpcUaClientNode: Status-Code:Good (0x00000000)

brudal1, 2022-05-04:
Hello,

after update to 0.2.275
i get at every trigger the Message "OpcUaClientNode: Status-Code:Good (0x00000000)" at the Debug.

How can i deactivate this message?

Any hints and links are welcome!

mikakaraila, 2022-05-04:
What is your current logLevel in the settings.js?
Change it to info or wanr or error.

// Configure the logging output
    logging: {
        // Only console logging is currently supported
        console: {
            // Level of logging to be recorded. Options are:
            // fatal - only those errors which make the application unusable should be recorded
            // error - record errors which are deemed fatal for a particular request + fatal errors
            // warn - record problems which are non fatal + errors + fatal errors
            // info - record information about the general running of the application + warn + error + fatal errors
            // debug - record information which is more verbose than info + info + warn + error + fatal errors
            // trace - record very detailed logging + debug + info + warn + error + fatal errors
            // off - turn off all logging (doesn't affect metrics or audit)
            level: "debug",
            // Whether or not to include metric events in the log output
            metrics: false,
            // Whether or not to include audit events in the log output
            audit: false
        }
    },

mikakaraila, 2022-05-04:
As I tested this quickly I didn´t get those messages even I had logging level debug.
So please be more specific and add more content when you will get the message and on the debug window or console?

brudal1, 2022-05-04:
i changed the setting in /data/settings.js under from "debug" to "error" and restarted NodRed.
Unfortunately I still get the message.

// Configure the logging output
    logging: {
        // Only console logging is currently supported
        console: {
            // Level of logging to be recorded. Options are:
            // fatal - only those errors which make the application unusable sh$
            // error - record errors which are deemed fatal for a particular re$
            // warn - record problems which are non fatal + errors + fatal erro$
            // info - record information about the general running of the appli$
            // debug - record information which is more verbose than info + inf$
            // trace - record very detailed logging + debug + info + warn + err$
            // off - turn off all logging (doesn't affect metrics or audit)
            **level: "error",**
            // Whether or not to include metric events in the log output
            metrics: false,
            // Whether or not to include audit events in the log output
            audit: false
        }
    },

i use NodeRed V2.2.2

[image]

It seems that the message is displayed even though the OPC UA note is disabled
[image]

I use 11 Flows with 17 OPC-UA_Client Notes.
I read Values from 7 different PLC's triggered by a Inject Note (Intervall 1 Sec)

mikakaraila, 2022-05-04:
And what is your logging level?

brudal1, 2022-05-04:
the logging level at the settings.js is on "error"

Exist a other config file for the logging level ?

mikakaraila, 2022-05-04:
What is actual action you are running? 

I tested with single read action and debug on the right is different to yours:
[image]

What is on the console? I expect you have something else wrong than just read response...

brudal1, 2022-05-04:
We are use allways "READ"
[image]

[image]

mikakaraila, 2022-05-05:
Is your inject msg for the read correct?

[image]

Test with one item first just to check it is working.

brudal1, 2022-05-05:
Yes, it works because I get values in the influxdb behind the OPC Client Note.

I cannot check this with the Message Note because there are too many debug messages with "Status-Code:Good".

[image]<https://user-images.githubusercontent.com/5184524/166873232-de72e767-b929-459b-af6f-10db2245d240.png>

Bitte beachten Sie, dass der Inhalt dieser E-Mail vertraulich zu behandeln ist. Sofern Sie nicht der beabsichtigte Adressat sind,dürfen wir Sie bitten, uns umgehend zu benachrichtigen und den Inhalt dieser E-Mail zu löschen. Vielen Dank!
Please ensure the confidentiality of this e-mail. If you are not the intended recipient, we would like to ask you to notify us immediately. Please delete the content of this e-mail in that case.Thank you!

ALPMA Alpenland Maschinenbau GmbH - Registergericht: Amtsgericht Traunstein, HRB 11720 - Geschäftsführer: Gisbert Strohn - Frank Eberle - Alpenstrasse 39-43 - 83543 Rott am Inn - Telefon: (0 80 39) 4 01-0 - Telefax: (0 80 39) 4 01-3 96 - E-Mail: ***@***.*** - www.alpma.de

brudal1, 2022-05-05:
to test i delete all flows and restart Node Red because i get the "Status-Code:Good" Message

now i have only one flow with a manual Trigger
and the OPC Read works
[image]

mikakaraila, 2022-05-05:
Good, add/expand and try to find out when it will start to produce Status-Code:Good messages.

brudal1, 2022-05-05:
I get the message "Status-Code:Good" with only one read action by one variable at each trigger

mikakaraila, 2022-05-05:
???

Attach that flow here so I can look and try to understand what is causing this.

brudal1, 2022-05-05:
[{"id":"4513ea1a8ad42715","type":"tab","label":"Flow 1","disabled":false,"info":"","env":[]},{"id":"bb088ba7811c4d8c","type":"OpcUa-Client","z":"4513ea1a8ad42715","endpoint":"89a16623.0422d8","action":"read","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","folderName4PKI":"","name":"","x":1920,"y":200,"wires":[["84e1e439eee51e08"]]},{"id":"120d046790b93f28","type":"inject","z":"4513ea1a8ad42715","name":"Trigger 1Sec","props":[],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":1270,"y":200,"wires":[["7cfd98b76bd079ee"]]},{"id":"84e1e439eee51e08","type":"debug","z":"4513ea1a8ad42715","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":2090,"y":260,"wires":[]},{"id":"7cfd98b76bd079ee","type":"change","z":"4513ea1a8ad42715","name":"V20ExMilkA\".\"Out\".\"Prod\".\"MilklineActive\"","rules":[{"t":"set","p":"topic","pt":"msg","to":"NodeID;ns=3;s=\"V20ExMilkA\".\"Out\".\"Prod\".\"MilklineActive\"","tot":"str"},{"t":"set","p":"measurement","pt":"msg","to":"MilklineActive","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":1600,"y":200,"wires":[["bb088ba7811c4d8c"]]},{"id":"89a16623.0422d8","type":"OpcUa-Endpoint","endpoint":"opc.tcp://10.10.10.11:4840","secpol":"None","secmode":"None","none":true,"login":false,"usercert":false,"usercertificate":"","userprivatekey":""}]

mikakaraila, 2022-05-05:
For some reason I got: 5 May 10:37:07 - [warn] Existing flow file is not valid json

brudal1, 2022-05-05:
`[{"id":"4513ea1a8ad42715","type":"tab","label":"Flow 1","disabled":false,"info":"","env":[]},{"id":"bb088ba7811c4d8c","type":"OpcUa-Client","z":"4513ea1a8ad42715","endpoint":"89a16623.0422d8","action":"read","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","folderName4PKI":"","name":"","x":1920,"y":200,"wires":[["84e1e439eee51e08"]]},{"id":"120d046790b93f28","type":"inject","z":"4513ea1a8ad42715","name":"Trigger 1Sec","props":[],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":1270,"y":200,"wires":[["7cfd98b76bd079ee"]]},{"id":"84e1e439eee51e08","type":"debug","z":"4513ea1a8ad42715","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":2090,"y":260,"wires":[]},{"id":"7cfd98b76bd079ee","type":"change","z":"4513ea1a8ad42715","name":"V20ExMilkA\".\"Out\".\"Prod\".\"MilklineActive\"","rules":[{"t":"set","p":"topic","pt":"msg","to":"NodeID;ns=3;s=\"V20ExMilkA\".\"Out\".\"Prod\".\"MilklineActive\"","tot":"str"},{"t":"set","p":"measurement","pt":"msg","to":"MilklineActive","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":1600,"y":200,"wires":[["bb088ba7811c4d8c"]]},{"id":"89a16623.0422d8","type":"OpcUa-Endpoint","endpoint":"opc.tcp://10.10.10.11:4840","secpol":"None","secmode":"None","none":true,"login":false,"usercert":false,"usercertificate":"","userprivatekey":""}]`

mikakaraila, 2022-05-05:
Please check X to Anonymous:
[image]

mikakaraila, 2022-05-05:
But anyway in my environment flow seems to work:
[image]

brudal1, 2022-05-10:
i enable Anonymous and save

But when i open again Anonymous is disabled

[image]

haph-vi, 2022-05-16:
I have the same problem but only on my linux machine (within Docker container, wait to confirmed by admin). On windows10 it doesn't occur. 
Even after deleting all nodes and deleting the OPC UA Config Node
[image]
and deploying, I get the same messages over and over.
I haven't restarted yet because I'm not admin on this server. 

This was my flow:
[
    {
        "id": "f6f2187d.f17ca8",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": ""
    },
    {
        "id": "f39cdba620c2b23e",
        "type": "OpcUa-Endpoint",
        "d": true,
        "endpoint": "opc.tcp://10.10.171.26:4840",
        "secpol": "None",
        "secmode": "None",
        "none": false,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    },
    {
        "id": "9f1f2a10.5552f",
        "type": "ui_group",
        "name": "Anomaly Dashboard",
        "tab": "8618be9a.b02898",
        "order": 1,
        "disp": true,
        "width": "22",
        "collapse": false,
        "className": ""
    },
    {
        "id": "8618be9a.b02898",
        "type": "ui_tab",
        "name": "Schweißspannung und Fehler",
        "icon": "dashboard",
        "disabled": false,
        "hidden": false
    },
    {
        "id": "49bc354ec69a3b40",
        "type": "ui_base",
        "theme": {
            "name": "theme-light",
            "lightTheme": {
                "default": "#0094CE",
                "baseColor": "#0094CE",
                "baseFont": "-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif",
                "edited": true,
                "reset": false
            },
            "darkTheme": {
                "default": "#097479",
                "baseColor": "#097479",
                "baseFont": "-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif",
                "edited": false
            },
            "customTheme": {
                "name": "Untitled Theme 1",
                "default": "#4B7930",
                "baseColor": "#4B7930",
                "baseFont": "-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif"
[... 1017 lines cut ...]
                "a6a559e5ab30ba10",
                "cb8b8fce2386c266"
            ]
        ]
    },
    {
        "id": "cb8b8fce2386c266",
        "type": "debug",
        "z": "f6f2187d.f17ca8",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 970,
        "y": 180,
        "wires": []
    }
]

A bit messy but was still developing.

brudal1, 2022-05-17:
i install V0.2.265.
With 0.2.265 works

haph-vi, 2022-05-17:
So downgrading from 0.2.275 helps? 

Anyway, just tested SUBSCRIBE instead of READ. This works and the error doesn't occur anymore. 

For my last comment: Restarting node-red stopped the "OpcUaClientNode: Status-Code:Good (0x00000000)" message.

mikakaraila, 2022-05-17:
I have modified code, but I will try to test more.... takes couple of days to have time to do this.

haph-vi, 2022-05-17:
OK!
I know this from my job too but often times I think I gave all information but it still misses some lol.

Do you need something more?

mikakaraila, 2022-05-17:
Just time to focus this, building new stuff at work. It is taking most of time now.

mikakaraila, 2022-05-20:
I think I figured out this one. New node-opcua contains improved keepalive and it is causing this.
It will send notifications that will cause status code good messages.

mikakaraila, 2022-05-23:
It would be good to read comments...

mikakaraila, 2022-05-24:
I change code a little bit, please test.

haph-vi, 2022-05-30:
Updated to 0.2.276, the status good message still comes in but only 1 time for every read action. 
Still spams the debug console if I read multiple values every 0.1s. I would like to be able to disable this message in the node settings.

mikakaraila, 2022-05-30:
I have fixed this. Wait next version.

Yusseiin, 2022-06-08:
Hi, may i ask if it his resolved in 0.2.278 or it will be fixed in next release?
On 0.2.278 still getting the same problem.

mikakaraila, 2022-06-08:
Then there is still something to fix, do you have test flow & output? Easier to fix if I can test it immediately.

Yusseiin, 2022-06-08:
`[{"id":"634167fbacd401eb","type":"tab","label":"Flow 4","disabled":false,"info":"","env":[]},{"id":"b99c592f85099a2e","type":"inject","z":"634167fbacd401eb","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":100,"y":80,"wires":[["9560d703f2b806c6"]]},{"id":"e71a6324224c1fe9","type":"debug","z":"634167fbacd401eb","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":670,"y":80,"wires":[]},{"id":"9560d703f2b806c6","type":"OpcUa-Item","z":"634167fbacd401eb","item":"ns=3;s=\"Data_block_1\".\"xBool\"","datatype":"Boolean","value":"","name":"","x":280,"y":80,"wires":[["11126b388786964c"]]},{"id":"11126b388786964c","type":"OpcUa-Client","z":"634167fbacd401eb","endpoint":"5205d044ad61d5d5","action":"read","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","folderName4PKI":"","name":"","x":500,"y":80,"wires":[["e71a6324224c1fe9"]]},{"id":"5205d044ad61d5d5","type":"OpcUa-Endpoint","endpoint":"opc.tcp://192.168.5.212:4840","secpol":"None","secmode":"None","none":true,"login":false,"usercert":false,"usercertificate":"","userprivatekey":""}]`

That's my code, do you need something more?

mikakaraila, 2022-06-08:
Thanks, found it!

Yusseiin, 2022-06-09:
Working flawless!!! Thanks a lot.

haph-vi, 2022-06-09:
For me too, thanks!

closed 2022-06-09 by mikakaraila