mikakaraila/node-red-contrib-opcua #435: OPC-UA Node doesn't restart after getting this error

AdemHamza, 2022-05-12:
Hello @mikakaraila ,

I have an issue with the opc-ua node for almost a week. I'm trying to connect to a OPC-UA server using a client node and the create a server on node red. I'm getting diffrent errors but mostly when the client goes to keep alive it doesn't restart when the server does.
for this time I got the Node-red server that crashed and I said maybe it would better if I opened an issue here :).
Node version: 0.2.257
I added the flow attached and a picture of the consol. hope it would Help
[Flow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/8677242/Flow.txt)
[image]

Thanks in advance

mikakaraila, 2022-05-12:
Hmm, cannot see first error it would tell me more what is happening. It is perhaps root cause for the problem.

AdemHamza, 2022-05-13:
Hello,
here's the full log.
[OPC-UA bug.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/8685255/OPC-UA.bug.txt)

mikakaraila, 2022-05-13:
First crash, remove variable if it will help:
cannot find private stuff for ActiveToolDataStructure
Error:
    at EXTENSION.get schema [as schema] (C:\Users\SESI009727\.node-red\node_modules\node-opcua-schemas\source\dynamic_extension_object.ts:387:25)
    at EXTENSION.toJSON (C:\Users\SESI009727\.node-red\node_modules\node-opcua-schemas\source\dynamic_extension_object.ts:686:37)
    at JSON.stringify (<anonymous>)
    at stringify (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\json-stringify-safe\stringify.js:5:15)
    at Object.encodeObject (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\util\lib\util.js:851:31)
    at sendDebug (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\nodes\core\common\21-debug.js:198:24)
    at DebugNode._inputCallback (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\nodes\core\common\21-debug.js:153:21)
    at C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:207:26
    at Object.trigger (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\util\lib\hooks.js:149:13)
    at DebugNode.Node._emitInput (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:199:11)
cannot find private stuff for MessageDataStructure
Error:
    at EXTENSION.get schema [as schema] (C:\Users\SESI009727\.node-red\node_modules\node-opcua-schemas\source\dynamic_extension_object.ts:387:25)
    at EXTENSION.toJSON (C:\Users\SESI009727\.node-red\node_modules\node-opcua-schemas\source\dynamic_extension_object.ts:686:37)
    at JSON.stringify (<anonymous>)
    at stringify (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\json-stringify-safe\stringify.js:5:15)
    at Object.encodeObject (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\util\lib\util.js:851:31)
    at sendDebug (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\nodes\core\common\21-debug.js:198:24)
    at DebugNode._inputCallback (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\nodes\core\common\21-debug.js:153:21)
    at C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:207:26
    at Object.trigger (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\util\lib\hooks.js:149:13)
    at DebugNode.Node._emitInput (C:\Users\SESI009727\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:199:11)
10 May 19:16:21 - [warn] [OpcUa-Client:MS085] MS085: Monitoring value: ns=2;s=1000070.VC1.Counters.GoodParts.ActualValue by interval of 5000 ms

One strange one:
11 May 04:45:00 - [error] [OpcUa-Client:MS085] Client node error on: MS085 error: [{}]

Later:
12:53:40.458Z :client_secure_channel_layer   :1354  ERROR !!! callback has thrown en error  Error: keepAliveManger already started
 ...                                                    at assert (C:\Users\SESI009727\.node-red\node_modules\node-opcua-assert\source\index.ts:12:21)
 ...                                                    at ClientSessionImpl.startKeepAliveManager (C:\Users\SESI009727\.node-red\node_modules\node-opcua-client\source\private\client_session_impl.ts:1914:15)
 ...                                                    at ClientSecureChannelLayer.<anonymous> (C:\Users\SESI009727\.node-red\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:1041:33)
 ...                                                    at modified_callback (C:\Users\SESI009727\.node-red\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:1352:31)
 ...                                                    at process_request_callback (C:\Users\SESI009727\.node-red\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:133:5)
 ...                                                    at ClientSecureChannelLayer._on_message_received (C:\Users\SESI009727\.node-red\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:788:9)
 ...                                                    at MessageBuilder.<anonymous> (C:\Users\SESI009727\.node-red\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:357:22)
 ...                                                    at MessageBuilder.emit (events.js:400:28)
 ...                                                    at MessageBuilder._decodeMessageBody (C:\Users\SESI009727\.node-red\node_modules\node-opcua-secure-channel\source\message_builder.ts:332:26)
 ...                                                    at MessageBuilder._feed_messageChunk (C:\Users\SESI009727\.node-red\node_modules\node-opcua-transport\source\message_builder_base.ts:252:18)

AdemHamza, 2022-05-20:
hello @mikakaraila ,
I did remedy to the last error but I got another one that I can't seem to figure out.
now the OPC-UA server Crashes (which is something that didn't hapen a lot before) and when it does the node red server crashes and to restart it I need to press enter in the cmd to restart the server (weirdly everything restarts without issues).
I restart the server 3 times a day at the begining of each shift.
I have a machine that inject values to the server using subscribe and the other is a modbus that injects values every 10 s.
Thank you in advance :)
[OPC-UA server issue.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/8737339/OPC-UA.server.issue.txt)

mikakaraila, 2022-05-20:
You are using ExtensionObject that client cannot detect and then decode.

cannot find private stuff for ActiveToolDataStructure

cannot find private stuff for MessageDataStructure
Error:
    at EXTENSION.get schema [as schema] (C:\Users\SESI009727\.node-red\node_modules\node-opcua-schemas\source\dynamic_extension_object.ts:387:25)
    at EXTENSION.toJSON (C:\Users\SESI009727\.node-red\node_modules\node-opcua-schemas\source\dynamic_extension_object.ts:686:37)
    at JSON.stringify (<anonymous>)

AdemHamza, 2022-05-20:
Hello @mikakaraila ,

Sorry to make you repeat yourself. but I thought I passed that error. So indeed it's a special type of nodes but I don't have the hand on the machine OPC server and I need those variables to track the production. is there a way to collect those data knowing that I tried to use TypeID to define the nodes.
Thanks

[image]
[image]
[image]

mikakaraila, 2022-05-20:
Client cannot access type: ns=2;i=3009 structure. It is on server Types. 
Even if you try to use Variable with Extension Object there will be still something...
The current implementation is made for basic types as if you look what is supported on the Item node.

mikakaraila, 2022-08-23:
Basic extension types should work, open again if this one needs some sub-types.

closed 2022-08-23 by mikakaraila