mikakaraila/node-red-contrib-opcua #437: OPC UA Server - how to set quality and timestamp of my variables/items?

gr-hayduk, 2022-05-24:
Hello OPC gurus,
I need to implement OPC UA Server in my nodered flow. I have managed to build node-hierarchy.xml and load it into the OPC UA server node (with custom nodeset directory). But then there's question about how to update variables (opc items) - not only the values, but also quality and timestamp?
Right now I have found only one possibility to update values only - to insert into the flow a OPC Client node and using this "backdoor" way, send item values to local OPC Server node. That solution, though, doesn't allow to set quality and the timestamp of last update (is that right?). This is what works for me now: 
[image]
_The background/application/intention:_ What I try to achieve is mqtt2opcua bridge, so when there's no mqtt update for longer time, I'd like to set opcua items quality to 'last usable value' and then after further timeout - 'bad, comm error'. It would be nice to be able to set not only quality but also the timestamp when the new quality was set.
Can anybody lead me to the correct path?
Thanks in advance
.

mikakaraila, 2022-05-24:
Variant contains only data type and value. DataValue contains variant, status code and sourceTimestamp and serverTimestamp.

gr-hayduk, 2022-05-24:
Thanks for quick reply, but I still don't know which Nodes in the flow I should use or what payload to use, to be able to set quality with the mentioned DataValue struct and OpcUaServer node. Can you explain me in more details, please.
Regards.

mikakaraila, 2022-05-24:
Almost done:
[image]

mikakaraila, 2022-06-05:
You can inject now status code and timestamp for the variable:

[image]

[image]

mikakaraila, 2022-06-05:
Quality value is from here:

Use actual decimal StatusCode value:
DataLost:                           2147483648
BadAggregateConfigurationRejected:  2161770496
etc.

Look values from Github node-opcua:
node-opcua/packages/node-opcua-status-code/source/_generated_status_codes.ts

gr-hayduk, 2022-06-05:
Thanks for notification and for your update, I'll try it tomorrow and get back with my feedback.

mikakaraila, 2022-06-06:
Thanks for the feedback, I was last week on vacation and it was quite late to test, commit and publish new version.
I just didn´t remember to add support for other nodeId formats.

I will merge your modifications to next version.

mikakaraila, 2022-06-06:
Merged, I will publish new version later....