mikakaraila/node-red-contrib-opcua #438: Payload coming after value is changing, But i need value for each second.

geekdk, 2022-05-30:
Hi Mika,
The payload is coming after the value changes even though I have subscribed to it for 1 second.
Here is the problem.
[image]

[image]

Please check the server timestamp as well as the subscribed time.

I know the value is not changing but I still want to read the same value.

Please let me know how to resolve it.

geekdk, 2022-05-30:
Thank for replying, 
I want to keep the subscription up, as I am reading multiple topics, and i want to keep the subscription up and not having much load on the OPC-UA server.

Do you think reading and injecting a new message every second is better than subscribing?

As I saw in the code, it is the group.on() changed the item only.
[image]

mikakaraila, 2022-05-30:
Reading is only way to get value every second as subscription is change
based.

It depends y server and how many values you are reading.

I have aggregator that can provide 100000 values in 700ms.

geekdk, 2022-05-31:
Thank you Mika, this may work

geekdk, 2022-05-31:
Hi Mika, When I am doing the reading for 10 tags, it only gives 4 at any given point in time.
Do you have anything about that in this regard? 
I am using 1 inject 10 items with a single opc-ua client on read mode.

geekdk, 2022-06-01:
Hi Mika, any suggestions?

mikakaraila, 2022-06-05:
New version published, please test.

closed 2022-11-10 by mikakaraila