mikakaraila/node-red-contrib-opcua #440: Can't browse subfolder

doteich, 2022-05-31:
I'm trying to browse the classic Siemens TIA OPC structure
[image]
However while browsing from Root further down I get stuck at the subfolder "DataBlocksGlobal" (ns=3;s=DataBlocksGlobal) as this only returns me a single Variable, while UA Expert shows all the available sub-objects. 
Even using the "native" Node.Js OPC UA package, im not able to get nodes from the subfolder

Only this: ` 
session.browse('ns=3;s=DataBlocksGlobal', (err, browseResult)=> {
if (!err) {                
       for(const reference of browseResult.references) {
            console.log( "   -> ", reference.toString());
        }
            }
            console.log(err)
        })
`
Returns: `   ->  { /*ReferenceDescription*/
 referenceTypeId               /* NodeId              */: HasProperty (ns=0;i=46)
 isForward                     /* UABoolean           */: true
 nodeId                        /* ExpandedNodeId      */: ns=3;i=5202
 browseName                    /* QualifiedName       */: Icon
 displayName                   /* LocalizedText       */: locale=null text=Icon
 nodeClass                     /* NodeClass           */: 2 /*(Variable)*/
 typeDefinition                /* ExpandedNodeId      */: ns=0;i=68
};
null`
If a go one step further and browse nodes in the subfolder, it works again. 
Is this a bug or is there any way to browse such kind of subfolders?

doteich, 2022-05-31:
@mikakaraila Thanks for your quick reply. As a workaround, is there a way to use the Crawler and just filter out nodes with NodeClass=Object? 
I'll open an issue in OPC UA repo as well.

mikakaraila, 2022-05-31:
There should be node that uses crawler and client node uses browse.

I can't remember without looking code.