mikakaraila/node-red-contrib-opcua #442: Can I use a custom server endpoint?

jptorrezan, 2022-06-16:
Hello, you have did a great work until here! Thank you!
I need to ask if there is a way to connect a client in a custom endPoint. I will get it from a configureFile, previous readed.
My application is used to collect data from different nodeIDs in different machines IPs, I get the IPs and Ports from a file, but i didn't find a way to input the endPoint in an OPC UA Client Node, I can just input the NodeId.

Thank you!
Joao Pedro

mikakaraila, 2022-06-17:
You can change port and resource path:
[image]

jptorrezan, 2022-06-20:
Oh, thank you.
I didnt know about this possibility, so let me ask another thing, all the fields in the OpcUaEnpoint object are filled with strings written like in the Edit Option, like:

[image]

mikakaraila, 2022-06-20:
Check this field (typo above): securityPolicy

jptorrezan, 2022-06-20:
Oh, sure!
So to read from an OPCUApath, in a custom server i will need make this? the field action is not very clear to me, it has the same values from an OPCUA Item Node?

[image]

mikakaraila, 2022-06-21:
I will update example for you. There are own actions for client / server node.

jptorrezan, 2022-06-22:
Thank you @mikakaraila , I'm having a problem when i try to connect with an user and password, I already get connected but without an user and password, can you say if my syntax is correct ?

[image]

Sorry for the insistence.

mikakaraila, 2022-06-22:
I tested and this one is working:

msg.topic="Reconnect";
msg.action="reconnect";
msg.OpcUaEndpoint = {
    credentials: {},
    endpoint: 'opc.tcp://localhost:53530/OPCUA/SimulationServer',
    securityPolicy: 'Basic256',
    securityMode: 'SignAndEncrypt',
    login: true,
    user: "test",
    password: "test"
}
return msg;

[image]

closed 2022-06-27 by jptorrezan