mikakaraila/node-red-contrib-opcua #443: Question: Configuration of Historical Access

AndreSchm, 2022-06-22:
Is there a way to configure the following two properties in context of Historical Access:

1. Storage of historical data
As far as I understood the default memory space for the historical data is the main memory.
Is there a way to set an other memory type like file system or a database?

2. Maximum data points per node in HA
The maximum amount of data points per node is capped at 1000.
Is there a way to increase this threshold? 
Reading [the docs](https://reference.opcfoundation.org/v104/Core/docs/Part11/5.4.2/) I found the _MaxReturnDataValues_ property. Is this implemented and configurable?

Thank you very much in advance.

mikakaraila, 2022-06-22:
1. Not at the moment, I didn´t put any options. There are some ways to add this feature.
   I will make small study how to add this. Any proposal what TSDB or database you would like to use?
2. See above, I could add options then user can define amount of data points.

I have implemented this for our product. It actually defines own class and uses it to make callbacks for the historian DB.

AndreSchm, 2022-06-23:
I would think of influxdb, postgres (maybe timescale) or MS sql server but I'm not sure, whether they are applicable for this purpose.

Eichi87, 2024-05-11:
Any news on implementing an option to increase buffer size and/or target an external database for persistent storage?