mikakaraila/node-red-contrib-opcua #444: Warning message "No session to close" after Deploy

Jacko00b, 2022-06-24:
Hi Mika !

We are using function node to inject endpoint to client node, and we inject it everytime that we deploy our project.
[image]
So, we get this warning for each client node (Write, Read and Subscribe):
[image]
Is there a way to correct it?

Thank you!

mikakaraila, 2022-06-27:
I don´t know your case context. It is just a warning and if you inject reconnect on start then it is normal that as long as there is no session (connection to any client then there is no session to close) you will get that warning.

closed 2022-07-02 by Jacko00b

Jacko00b, 2022-07-19:
Sorry to re-open it but we still have the problem

For the context, we reconnect to server every time that we deploy our project so we can switch adresses whenever we want
[image]
And in opcua route we are setting our endpoint
[image]

Should we just " connect " to the Opcua-Client node ?

Thanks!

mikakaraila, 2022-07-20:
There is no action for "connect". So you have to use "reconnect".