mikakaraila/node-red-contrib-opcua #448: ocpua event node don't send the node id

Clarisse65, 2022-06-29:
I want to aknowledge alarms after it is receive when i click on it but the opcua event don't send the node id that is need to aknowledge the alarm.

mikakaraila, 2022-06-29:
I have example to acknowledge alarm, I will add it here... have to find it first.

mikakaraila, 2022-06-29:
Look examples and OPCUA-TEST-NODES.json this specific part:

[image]

Function node contains object and condition to acknowledge:
[image]

Client node action is just acknowledge:
[image]

Clarisse65, 2022-06-29:
Thank you for your answer,
In your example i don't see that you use an information that opcua event and opcua client node give you to acknowledge the alarm.  To do it, i should take the ns=...;s=... and event id informations from the opcua event and opcua client node msg ? but i am not seeing these informations ?
Do i don't understand something ?

mikakaraila, 2022-06-29:
msg.topic contain actual nodeId == Object/Variable in this case Prosys UA Simulation server Alarm.
msg.conditionId is actual branch (Latest event) Id that is needed to acknowledge alarm
msg.comment is just string that server will store inside alarm object comment variable

NodeId is OPC UA variable address, it contains namespace index ns= part in the NodeId and then address:
 s=String addressing 
 i=Index address
 b = opaque (hex) address kind of guid

Clarisse65, 2022-06-29:
I understand. But how can i found these information in the msg send by the opcua event node ?

mikakaraila, 2022-06-29:
SourceName in case of namespace index is know, otherwise ConditionId:

[image]

mikakaraila, 2022-06-29:
Just checking code, it should return SourceNode as it receives actual nodeId in that field, but it is not visible in the payload.

mikakaraila, 2022-06-29:
I will actually set msg.topic = msg.payload.SourceNode;

This will help to use it to acknowledge alarm.

mikakaraila, 2022-06-29:
New version 0.2.282 published with fixes, please test.

Clarisse65, 2022-06-30:
I tested and the sourceid in the msg.topic is still not the information needed to aknowledge the alarm for me...
I don't know how to explain...
I have an TriggerEmergencyStop that fire the AlarmEmergencyStop. I need "ns=2;s=AlarmEmergencyStop" to aknowledge the alarm but in the topic i receive "ns=2;s=TriggerEmergencyStop". Is that normal ?

mikakaraila, 2022-06-30:
Can you test acknowledge with UaExpert?
Take a screenshot from the event viewer alarm details are interesting. 
If alarm payload contains needed information you have to use function node to use correct nodeId.
Which server is used?

Clarisse65, 2022-07-01:
Maybe i found something, if i am wrong i will send you some screenshot. For me, the information needed to aknowledge alarm is called "ConditionId" in the opcua event node but i don't receive it in the opcua client node in event mode. I found out this "ConditionId" have attribute=1, ns=2782 and browsepath empty. So i tried to take it with an function node between the opcua event node and the opcua client node :

msg.eventFilter.selectClauses[11].typeDefinitionId = 
{identifierType : 1,namespace : 0, value : 2782};
msg.eventFilter.selectClauses[11].browsePath = [];
msg.eventFilter.selectClauses[11].attributeId = 1;
return msg;

Almost worked... but i receive the same alarms  various times and the condition id don't show in all... 
I send youscreenshots of  the msg i receive after the function i wrote and of the msg i receive after the opcua client node  where the ConditionId show up.

Tell me if my explainations are not clear and tell me what you think about it please !

msg after my function :

![capture1](https://user-images.githubusercontent.com/108391020/176919374-392653b1
[image]
-3b8a-4d00-9786-c17753d3767e.PNG)

msg after opcua client node : 

[image]

Clarisse65, 2022-07-01:
First screenshot didn't work sorry, i send it again 👍 
[image]

mikakaraila, 2022-07-03:
OK, I was asking UaExpert view, but never mind. 

My mistake as aknowledge "uses" ConditionId" it should be used instead of SourceNode.
I will take a look tomorrow...

Clarisse65, 2022-07-04:
If it's not to late here is some pictures of uaexpert but i am not sure that is the view you wanted : 
[image]

[image]

[image]

[image]

[image]

mikakaraila, 2022-07-04:
Hmm, with Prosys UA Simultation server it is not providing ConditionId.
Just wonder if it should be read by using SourceNode -> Alarm object -> ConditionId.

This will take some time, I will try to find it from the specifications, but normally they don´t contain practical examples for using the data.

It can be also that ConditionId depends on alarm type, so I cannot test it with my test server.
As in your debug output it is in the payload. Perhaps I will not update topic inside my code.
Users must pick up SourceNode/ConditionId depending the alarm type...

Koseng, 2022-08-07:
Hi guys,
just to let you know, I am fixing it. I've got the ConditionId already working. Wanted to finish, but found another bug regarding the events which I need to fix. More info in the next days.

mikakaraila, 2022-08-08:
Excellent, thanks!
I have been on vacation and last week full of normal work load.

Koseng, 2022-08-09:
Created now the PR #465 .

The general problem was a wrong entry in the SelectClause of the EventFilter. This is the most tricky thing of the whole Alarms & Conditions :-). The Operand for the ConditionId must be a special operand. It is different to all the others.

ConditionId must be added with a special operand in the SelectClause of the EventFilter. The ConditionId is not a standard parameter of the ConditionType. The ConditionId is the NodeId of the instance of the ConditionType. It is something special and therefore the Operand in the SelectClause is different. See different Operand at bottom of https://reference.opcfoundation.org/v104/Core/docs/Part9/5.5.2/.

Further necessary improvements for Events in nodered:
- In the current Events node, all the fields of an EventType should be collected automatically and the EventFilter should be created automatically. There should be no manual coding of fields. It should be done like in https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client/source/alarms_and_conditions/client_alarm_tools_extractConditionFields.ts
- I think it would be good to add a new CustomEvent node, where the EventType and the Fields are all entered manually in the NodeRed UI of that node.

closed 2022-08-10 by mikakaraila

mikakaraila, 2022-08-10:
Thanks, merge done. I will update node-opcua and fix/check extension/custom object cases.
Then I will publish new version.

After that I will take a look how to improve and add custom event fields (there was one lady from Siemens who did it, but I cannot find code).

Koseng, 2022-08-10:
@mikakaraila 
Before you create a new version:
I'm still wondering whether it is a good idea to use a different field for the msg.topic for different event types. I think we should not use the ConditionId as msg.topic, but always use EventType as msg.topic.

Also I'm currently working on creating the EventFilter fully automic as mentioned above. A lot of stuff is already working.

mikakaraila, 2022-08-10:
It's actually needed to acknowledge alarm.

I already published new version...

Koseng, 2022-08-10:
Ok i see, then the Acknowledge interface of the client also would need to be changed.

mikakaraila, 2022-08-11:
Think twice, it will break existing flows. Purpose with flow programming will enable connecting nodes to make smooth application implementation without programming.

I mean node with event subscription -> output msg -> filter/delay -> acknowledge wanted alarms.

calumk, 2023-02-09:
@mikakaraila 

Do you have an exmple of acknowledgeing the alarm - Im quite stuck on how to achieve this.

mikakaraila, 2023-02-10:
Read more carefully from the start. Example is included in the OPCUA-TEST-NODES.json

[image]

calumk, 2023-02-10:
Hi Mika, Thanks for your response - everything has been read - but we are still experiancing some issues.

The Flow example you give, shows the acknowledgement of an event, but you hard-code the conditionId etc in the function.
The issue is, that we dont seem to be able to get this data, in order to pass it back to the acknowledge.

In other words, 
The event we subscribe to recieves the alerts, but does not seem to pass the required data to acknowledge them.

We are using a Siemens ET200SP plc, with OPC-UA alarms - it is possible it generates a different format? (We can see the conditionId in UA Expert, but it doesnt come to node-red.

The second issue, is that setting the mode to "acknowledge" in the opcua-client, and then passing it a message (of any sort) or the function as you have it in the demo, is causing node-red to crash immediatly. - I dont have the error log to hand at the moment, but it is something like. "Cannot convert null to string"  - Essentially it seems that some property is missing or malformed, but we are unable to track down the cause of this.

mikakaraila, 2023-02-10:
Ok, you will need to subscribe events and then use filter / collect wanted events that you want to acknowledge.
I will check if I have time to make an example how to do this as flow... not today, but during weekend / next week.

mikakaraila, 2023-02-10:
Enough time to fix to get it working more easily:

[image]

As event comes also as it has been acknowledged status will show it:
[image]

calumk, 2023-02-10:
This  looks like it will be very helpful -  thanks

Will you push an example?

mikakaraila, 2023-02-12:
Here is example:
[AUTO-ACKNOWLEDGE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10715569/AUTO-ACKNOWLEDGE.txt)

PR #465 (2022-08-10): Fix ConditionId for Events
Fixes #448 

- Removed the basically unused getBrowseName and tried to make the code more understandable
- If no ConditionId exists, use EventType as topic. A SourceNode can be source of different events. So I think the EventType is the most interesting property for an event. 
- ConditionId is added with a special operand in the SelectClause of the EventFilter. The ConditionId is not a standard parameter of the conditionType. The ConditionId is the NodeId of the instance of the ConditionType. It is something special and therefore the Operand in the SelectClause is di [... cut]