mikakaraila/node-red-contrib-opcua #450: Mute OPC-UA read status

subwayfootlong, 2022-07-06:
Hello all,
I am using the OPCUA Node to subscribe to multiple nodeIds and upon a good read, it will output: CytroPac: StatusCode: Good (0x00000000)
[image]
Usually this would be ok with the debug filter however, it is also clogging up the terminal which I use to monitor my database input.
[image]
Is there a way for me to mute that node so that it would not send out debugs?

mikakaraila, 2022-07-06:
Similar case was fixed lately, are you using latest version?

subwayfootlong, 2022-07-08:
@mikakaraila ahhh thanks just updated it thank you

closed 2022-07-08 by subwayfootlong