mikakaraila/node-red-contrib-opcua #451: OPC UA Server Node and issuing/creating Certificates

youngkzy, 2022-07-08:
QUESTION: 
  I have used Kepwarer to issue Trusted Server and Trusted Client Certificates.   Lets say I do not have this tool to do so. Can you please go over the "Best Practice" for issuing Trusted Server and then Trusted Client Certificates to be used on Client and Server?   What software or process works?  What would you do?  And issuing them with a long end date period so to limit expiring Certificates? 
Thanks,

mikakaraila, 2022-07-08:
If server don´t accept anonymous user and access level & permissions are configured that only users with the suitable role can write value / execute methods then I expect there is some certificate renewal interval 1-2 years or even more good practise.

This will keep following:
1) Only trusted clients with strong certificates can access server
2) Access level will control general level read / write access to variables
3) Permissions can be used to allow write / methods execution to only selected users with the role like operator / engineer
    See WellKnownRoles from the OPC UA specification: https://reference.opcfoundation.org/v104/Core/docs/Part3/4.8.2/

youngkzy, 2022-07-11:
I'm pursuing UAExpert to use on Linux and Windows. I was also wondering about Kleopatra Certificate manager to see if you can use it to manage the OPC UA Certifications. I would like to understand better the usage of Certificates vs. 'None' when it comes to  Certificate and signing. Kepware is pretty good to go over this topic and additionally Anonymous vs. Login/Password for security. No need to explain Certificates, but just if you can use any tool, Like Kepware, UAExpert to build/Create and manage them. Kepware and Windows is pretty straight forward, But Linux cannot run Kepware.  Do you know of any educational documents or otherwise that goes over this OPC UA topic?  Thanks

mikakaraila, 2022-07-12:
Look node-opcua and certificate manager. It "automates" a lot and Etienne is adding certificate expiration alarm.

It is new feature to get notification (alarm) before certificate has expired.

mikakaraila, 2023-07-04:
Latest version is not anymore using openssl, instead internal crypto packages.