mikakaraila/node-red-contrib-opcua #452: Crawler: Cannot find reference in cache

geekdk, 2022-07-08:
I am getting this error which crashes the whole node-red server,

8 Jul 07:29:04 - [red] Uncaught Exception:
8 Jul 07:29:04 - [error] Error
    at assert (/data/node_modules/node-opcua-assert/dist/index.js:11:21)
    at /data/node_modules/node-opcua-client-crawler/dist/node_crawler.js:47:48
    at /data/node_modules/node-opcua-client-crawler/dist/node_crawler_base.js:240:17
    at /data/node_modules/async/dist/async.js:1496:46
    at Array.forEach (<anonymous>)
    at trigger (/data/node_modules/async/dist/async.js:1496:27)
    at /data/node_modules/async/dist/async.js:1569:21
    at /data/node_modules/async/dist/async.js:327:20
    at /data/node_modules/node-opcua-client-crawler/dist/node_crawler_base.js:149:21
    at /data/node_modules/node-opcua-client-crawler/dist/node_crawler_base.js:415:24
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-red-docker@2.2.2 start: `node $NODE_OPTIONS node_modules/node-red/red.js $FLOWS "--userDir" "/data"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-red-docker@2.2.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /data/.npm/_logs/2022-07-08T07_29_04_296Z-debug.log

Image : [image]

Debug Log : 
[image]

mikakaraila, 2022-07-08:
Seems to be something internal in the node-opcua. 
As error message states contact sterfive support: https://support.sterfive.com/

geekdk, 2022-07-11:
Thanks @mikakaraila, I solved the problem, it seems like it was all getting caught at caches and it was trying to read by that even though it was cleaned up after redeploying.
I have tweaked the code and am able to manage the situation.

Thanks for your prompt response.