mikakaraila/node-red-contrib-opcua #453: Non recursive browsing

ridort, 2022-07-11:
Hello,
Is it possible to browse one node without crawling the whole subtree?
I just want to find out what are the children of a particular node and stop there.

mikakaraila, 2022-07-12:
Use action "INFO"

ridort, 2022-07-12:
Hi, the INFO action doesn't return the child nodes (references) of the queried node like the BROWSE action does. The `references` and `parent` info returned by BROWSE action is what I want to see, without also returning the browse result of every other child, child of child, etc. in the subtree.

INFO

[image]

BROWSE

[image]

mikakaraila, 2022-07-13:
Only solution at the moment is BROWSE, sorry.

For the root node ns=0;i=85 there is not parent. You can try to filter result payload or use function node to get only the part of the subtree you want to use.

closed 2023-06-28 by mikakaraila