mikakaraila/node-red-contrib-opcua #454: WRITE MULTIPLE option in client not working correct

rugee-f, 2022-07-11:
Hi Mika, I have a problem with the WRITE MULTIPLE function of the OPC UA client node.

I used this function in an older version (0.2.269 on Node-Red 1.x.y) without any problem.
The msg.topic is set to "ALL" and in the msg.payload I pass the array with my items 

items[0] = {
            "nodeId":cam_ldt_ts.Item,
            "datatype":cam_ldt_ts.datatype,
            "value" : new Date(msg.payload),
            };

to the WRITE MULTIPLE client node.

Now I use the same flow in a new installation (Node-Red 2.2.2 + OPC UA nodes Version 0.2.282) 
and I got an error by running the flow:

Client node error on: TXT Sensor Data (ALL) error: ["Session is not active!"]

Is it possible that something was changed from version 0.2.269 to 0.2.275? 
I tested my flow on several Node-Red installations I have available
and found that since 0.2.275 the WRITE MULTIPLE behaves different.

I also stepped up Node-Red to version 2.x.y in that time,
so the last working WRITE MULTIPLE version was on Node-Red 1.x.y.
Is that possible?

mikakaraila, 2022-07-12:
Topic "ALL" is not used anymore, write multiple action will store item, in case of topic is "writemultiple" it will write items.
And in case of "clearitems" it will remove all items.

```Javascript
        // Store nodeId to read multipleItems array
        if (msg.topic !== "writemultiple" && msg.topic !== "clearitems") {
```

rugee-f, 2022-07-13:
Hi Mika, is there a working example somewhere in the documentation?

I am not sure how I should prepare the client node to store the nodeIds.
I changed my Function Node code tu use topic "writemultiple"

[FnNodeCode Start]
var new_msg = {};
var items = [];
var msgs = [];

//read global vars
//camera
var cam_ldt_ts=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@CameraPicture@ldt_ts')
var cam_s_data=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@CameraPicture@s_data')

//bme680
var env_i_aq=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@EnvironmentSensor@i_aq')
var env_di_gr=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@EnvironmentSensor@di_gr')
var env_r_h=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@EnvironmentSensor@r_h')
var env_i_iaq=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@EnvironmentSensor@i_iaq')
var env_r_p=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@EnvironmentSensor@r_p')
var env_r_rh=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@EnvironmentSensor@r_rh')
var env_r_rt=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@EnvironmentSensor@r_rt')
var env_r_t=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@EnvironmentSensor@r_t')
var env_ldt_ts=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@EnvironmentSensor@ldt_ts')

//ldr
var bri_ldt_ts=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@BrightnessSensor@ldt_ts')
var bri_r_br=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@BrightnessSensor@r_br')
var bri_i_ldr=global.get('@ns=3;s=gtyp_Interface_Dashboard@Subscribe@BrightnessSensor@i_ldr')

items[0] = {
            "nodeId":cam_ldt_ts.Item,
            "datatype":cam_ldt_ts.datatype,
            "value" : new Date(cam_ldt_ts.Value),
            };

items[1] = {
            "nodeId":cam_s_data.Item,
            "datatype":cam_s_data.datatype,
            "value" : cam_s_data.Value,
            //"value" : 1,
            };

//bme680
items[2] = {
            "nodeId":env_i_aq.Item,
            "datatype":env_i_aq.datatype,
            "value" : env_i_aq.Value,
            };

items[3] = {
            "nodeId":env_di_gr.Item,
            "datatype":env_di_gr.datatype,
            "value" : env_di_gr.Value,
            };
items[4] = {
            "nodeId":env_r_h.Item,
            "datatype":env_r_h.datatype,
            "value" : env_r_h.Value,
            };
items[5] = {
            "nodeId":env_i_iaq.Item,
            "datatype":env_i_iaq.datatype,
            "value" : env_i_iaq.Value,
            };
items[6] = {
            "nodeId":env_r_p.Item,
            "datatype":env_r_p.datatype,
            "value" : env_r_p.Value,
            };
items[7] = {
            "nodeId":env_r_rh.Item,
            "datatype":env_r_rh.datatype,
            "value" : env_r_rh.Value,
            };
items[8] = {
            "nodeId":env_r_rt.Item,
            "datatype":env_r_rt.datatype,
            "value" : env_r_rt.Value,
            };
items[9] = {
            "nodeId":env_r_t.Item,
            "datatype":env_r_t.datatype,
            "value" : env_r_t.Value,
            };
items[10] = {
            "nodeId":env_ldt_ts.Item,
            "datatype":env_ldt_ts.datatype,
            "value" : new Date(env_ldt_ts.Value),
            };
//ldr
items[11] = {
            "nodeId":bri_ldt_ts.Item,
            "datatype":bri_ldt_ts.datatype,
            "value" : new Date(bri_ldt_ts.Value),
            };
items[12] = {
            "nodeId":bri_r_br.Item,
            "datatype":bri_r_br.datatype,
            "value" : bri_r_br.Value,
            };
items[13] = {
            "nodeId":bri_i_ldr.Item,
            "datatype":bri_i_ldr.datatype,
            "value" : bri_i_ldr.Value,
            };

new_msg.topic ="writemultiple";
new_msg.payload = items;

return new_msg;
[FnNodeCode End]

 but I get the error message: 
Client node error on: TXT Sensor Data (ALL) error: ["TXT Sensor Data (ALL) no items to write"]

mikakaraila, 2022-07-13:
First add items and then call with the topic writeall, see example ALL function node:

msg.payload = [];

msg.topic="ns=5;s=Double";

msg.payload[0] = 
{
    nodeId: "ns=5;s=Double",
    datatype: "Double",
    value: 22.33
};

msg.payload[1] = 
{
    nodeId: "ns=5;s=Int32",
    datatype: "Int32",
    value: 32
};

return msg;

rugee-f, 2022-07-14:
[flows_Multiple.json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9112040/flows_Multiple.json.txt)

rugee-f, 2022-07-14:
Sorry my explanation was not correct.
hitting the [Add ALL] inject from your example will throw that red marked error but the signals are send immediately.
When the value doesnt change nothing is sent. If I reset your variables they are send agian.
But it is not in that way, that you have to load NodeIDs and have to trigger in a second shot with topic "writemultiple".
This is only in the case that you fill NodeIDs one by one...

mikakaraila, 2022-07-15:
Hmm, I tested your flow and it is working.

There is something wrong in your function node "array to write":

[image]

mikakaraila, 2022-07-15:
Perhaps documentation / example is unclear, but as payload is Array and topic is NOT writemultiple / clearitems code will
write payload array. **Code will not store values**. New payload array with new values can be used to write again new values.

This was wanted a long time ago.

If payload is just variable item then it will be stored and topic "writemultiple" will be used as command to write stored values.
Topic "clearitems" can be used to clear stored values.

rugee-f, 2022-07-16:
Hi Mika, I had a mistake in my code where I build the topic names for subscribing the signals. 
That was the reason why you get the error by "array to write".
I had to adjust the example that the used environment variables are created automatically in the demo and did something wrong. In my app I do this from a file...  So I reworked my example and got still some more questions:

When I do the "one by one" action with the UA Item nodes I see differences in my Flow and in your example.
Your Inject node:

[image]

with this UA Item node:

[image]

is causing this: 

[image]
 

My example 
Inject:
[image]

UA Item:
[image]

Result:
[image]

Why?

rugee-f, 2022-07-16:
I also found another problem with the Inject node before the UA Item:

Inject:
[image]

Item:
[image]

is building this message (strangely there is now yellow warning coming up): 

[image]

by hitting "write All" the value 0 is send to the UA Item not the 2 that was setup in UA Item node

[image]

This seems to be an issue of the Inject node where payload is setup to be numeric.

[image]

rugee-f, 2022-07-16:
And at last I like to know if you get that red message (even in your example) when you hit "Add ALL"

I see that the writing is taking place (Subscription is creating green marked message) but error message is also present...

[image]

rugee-f, 2022-07-16:
[flows_Test_rework.json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9125771/flows_Test_rework.json.txt)

reworked version

mikakaraila, 2022-07-17:
Simple case as item node has value as String it will be converted correctly to variant value.
I expect build variant cannot convert Date() to DateTime.  
Node-opcua will check datatype & value are valid.

Session case was fault in case of the other writeall was used to write values. 
So there was mistake in the logic.

mikakaraila, 2022-07-19:
New version available, please test.

rugee-f, 2022-07-20:
Hi Mika, the basic functionality seems to be OK, but I still have an (maybe cosmetic) issue:

The datatype of the Inject node is effecting what the OPC UA Item node does.
I think this is not really clear for the user.
If the value field of UA Item node is filled you should use this value with the proper datatype.
If it is not filled you can use the payload of leading node (in this case Inject) to fill the UA Items node value.
I created a new example flow where you can see what I mean.
1. case:  -Inject paylod type string -UA item value filled with '2' - datatype INT16 -> 2 is written to UA client node
[image]

2. case:  -Inject paylod type number -UA item value filled with '21' - datatype INT16 -> 0 is written to UA client node!
[image]

3. case:  -Inject paylod type string -UA item value filled with 'abc' - datatype String -> abc is written to UA client node
[image]

4. case:  -Inject paylod type number -UA item value filled with 'START' - datatype String -> something is written to UA client node but is causing an error.
[image]

Maybe it is more clear for the user that UA item nodes value is used whenever it is filled.
If value is empty you can use the payload of leading node to generate the passed message.
If the datatype is wrong there it is ok to throw an error.

New Test Flow:
 [flows_Test_rework_2.json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9147942/flows_Test_rework_2.json.txt)

mikakaraila, 2022-07-20:
After my vacation, not today. I will study what can be improved.

mikakaraila, 2023-09-13:
Did you test this again. Making cleanup, closing this one. Reopen if not working.

closed 2023-09-13 by mikakaraila