mikakaraila/node-red-contrib-opcua #455: [Feature Request] Endpoint configuration options: MaxMessageSize, MaxBufferSize

pascalporedda, 2022-07-12:
I'm currently trying to integrate an OPC endpoint into my flow, that offers a method, which returns quite a large message, which overflows the default message size. 
In the UA Expert client, I had to increase my MaxMessageSize, in order to not run into a timeout / error. 

The node-red library offers configuration options for these properties, see: https://node-opcua.github.io/api_doc/2.0.0/interfaces/endpointconfigurationoptions.html#maxmessagesize

I assume that it should be relatively easy to make this configurable in the OPC nodes as well? I will have a look into this, if I do find a quick way I'll open up a PR if that is in your interest as well?

mikakaraila, 2022-07-12:
Small change in server html & JavaScript should make it possible.

mikakaraila, 2022-07-12:
I will add fields now, it is just 2 fields in the options.

pascalporedda, 2022-07-12:
Awesome, thank you! You're fast. 
I was still trying to figure out how the node-opc API works, got to the point that there is a transportSettings property for the connection options

mikakaraila, 2022-07-12:
Just tested that with default value 4096 still working. I will publish this now.

mikakaraila, 2022-07-12:
New version available, please test.

pascalporedda, 2022-07-12:
Your changes reflect the server node, but what about the client? That's what I'm fiddling around with. 
According to: https://github.com/node-opcua/node-opcua/blob/45a8e4c223d5f5555d742103e1e9930152157548/packages/node-opcua-client/source/client_base.ts

There is a transportSettings property, that allows setting the maxMessageSize in the client. I have tried setting that in the setupClient function, but so far no luck with getting it to work.

pascalporedda, 2022-07-12:
So I've digged this down a little bit:

There is the option to set it in the client, but node-opcua does not respect this setting. I will open up an issue there and link it here.

mikakaraila, 2022-07-13:
I expect both server and client must support MaxMessageSize and MaxBufferSize. I will update client options.

AlbertoPilatoDerga, 2023-04-21:
Hi @mikakaraila,
can you please advise on how to specify maxMessageSize parameter when connecting as a client to an external OPC-UA endpoint?

I cannot find it in the OpcUa-Endpoint configuration node. 
I tried injecting msg.maxMessageSize or msg.MaxMessageSize, but it doesn't seem to work.

Thanks,
A

mikakaraila, 2023-04-23:
It can be set when you create client. Example can be found:
https://github.com/node-opcua/node-opcua/blob/156520cbcb95c0f059ffa1aff743cc23b29ae95c/packages/node-opcua-samples/bin/tiny_client.ts

I haven´t added as a parameter as normally default size has been working.

AlbertoPilatoDerga, 2023-05-02:
Thanks @mikakaraila ,
I saw the options in the server settings, would you mind adding the same to the client?

I did not understand how I should configure my client in Node-RED to set the MaxMessageSize. I already managed to set it working on a test NodeJS client as the one you shared, but I need to be able to use it in a Node-RED flow.

mikakaraila, 2023-05-02:
Ack, some lines and node UI must be updated. I will take a look today evening or tomorrow...

mikakaraila, 2023-05-02:
Published new version, try to test it! Works with default (empty values).

AlbertoPilatoDerga, 2023-05-03:
Great @mikakaraila , which version is it? 0.2.304 seems latest one available

mikakaraila, 2023-05-03:
Should be v0.2.305

https://flows.nodered.org/node/node-red-contrib-opcua

AlbertoPilatoDerga, 2023-05-03:
I updated the palette and tested the endpoint. it now works perfectly.