mikakaraila/node-red-contrib-opcua #457: OPCUA-Variable.json set variable doesn´t work

SteigiMa, 2022-07-19:
Hello, i tried the OPCUA-Variable.json. The Server is running and communicates with opc ua expert, I can set Folders and Variables. But
I can´t set a value. Is there something I still need to consider, or can anybody explain how this is working? Any help is greatly appreciated. Thank you

mikakaraila, 2022-07-20:
Here is one example
[OPCUA-VARIABLE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9146755/OPCUA-VARIABLE.txt)
.

SteigiMa, 2022-07-20:
Thank you for you reply. I tried this our but it is the same problem, i can see the variable (addTestVariable) in UA Expert but if I try to set a value (Set TestVariable false) I didn´t see a change in UA Expert.

SteigiMa, 2022-07-20:
Here the output of the debug when I set the variable

msg.payload : Object
{ messageType: "Variable", variableName: "TestVariable", datatype: "Boolean", namespace: "1", variableValue: false }

mikakaraila, 2022-07-20:
I am still on vacation, but it would help if you attach your flow here. 

As my example is working you have misunderstood something...

SteigiMa, 2022-07-21:
I tried with the latest version of "node-red" and "node-red-contrib-opcua", it works very well! It seems that in the older Version (Node Red 2.0.6, OPC ,0.2.236) this function is not working. Thank you for your support and have great vacation.

closed 2022-07-21 by mikakaraila