mikakaraila/node-red-contrib-opcua #458: Error while writing / reading Extension Objects

Verschnupft, 2022-07-26:
Hello, I am trying to write / read variables (UDT's) from a Siemens PLC. It works for the most part but I always run into an Error while reading / building the Extension Object. Writing values works fine.
The correct values are displayed while reading and the values are changed correctly.
Do you have an explanation or idea how to fix this?

I'll try to add every necessary file.
the Flow:
flow
[Flow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9165829/Flow.txt)

the Errors:
[Error.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9165875/Error.txt)

the Datatype:
TIA Portal
[image]

UAExpert
[image]

mikakaraila, 2022-07-26:
I will take a look next week, perhaps updating node-opcua will solve this. You can take a try...

mikakaraila, 2022-07-28:
We have to wait this one will be solved: https://github.com/node-opcua/node-opcua/issues/1174

mikakaraila, 2022-08-10:
Please test, I updated dependency node-opcua v2.73.0 that contains fixes related to this.

Verschnupft, 2022-08-11:
Unfortunately I am still getting the same error. 
Current versions:
Node: 16.14.0
Node Red: 3.0.2
node-red-contrib-opcua: 0.2.285 (this is the build number of the current version, right?)
TIA: V17

The error only occurs after building/reading/writing for the first time. If I try to do one of those actions for a second time, everything works fine. 

It's the same error as before with these nodeId's: 
- ns=2, i=6522
- ns=2, i=6525
- ns=3, i=3400
- ns=3, i=3500
- ns=2, i=15889
- ns=2, i=15888

mikakaraila, 2022-08-11:
Hmm, I don´t have TIA so it will be hard to fix this, but I will try to find bug without debugging.

But as you comment second time everything works? Can you make test with one nodeId and run it twice as it will work.
Settings debug on and console output could help me to fix this without TIA (I hope).

Verschnupft, 2022-08-11:
Thank you for your fast replies!
I'll document my tests in the following text and I'll add .txt documents with the same text:

[image]

[Flow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9306753/Flow.txt)

[Test1_build.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9306764/Test1_build.txt)
[Test2_read.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9306765/Test2_read.txt)
[Test3_write.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9306771/Test3_write.txt)

### Test 1 - building the object (nodeId: ns=3;s=DT_"DataFromDb")
first time:
`11 Aug 08:21:35 - [info] Started flows
11 Aug 08:21:35 - [info] [mqtt-broker:a62fd28ff9712b7a] Connected to broker: mqtt://localhost:1883
06:21:44.544Z :populate_data_type_manager_104:38    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6522
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:31:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:21:44.572Z :populate_data_type_manager_104:167   err= Error
 ...                                                    at assert (C:\Users\Install\.node-red\node_modules\node-opcua-assert\dist\index.js:11:21)
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:164:52
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:21:44.614Z :populate_data_type_manager_104:38    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6525
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:31:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:21:44.644Z :populate_data_type_manager_104:167   err= Error
 ...                                                    at assert (C:\Users\Install\.node-red\node_modules\node-opcua-assert\dist\index.js:11:21)
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:164:52
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:21:44.682Z :populate_data_type_manager_104:38    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3400
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:31:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:21:44.696Z :populate_data_type_manager_104:167   err= Error
 ...                                                    at assert (C:\Users\Install\.node-red\node_modules\node-opcua-assert\dist\index.js:11:21)
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:164:52
[... cut ...]
11 Aug 08:51:06 - [info] [debug:debug WRITE]
{
  _msgid: '11c497dd9cb30c60',
  payload: ConstantStatusCode {
    _value: 0,
    _description: 'The operation succeeded.',
    _name: 'Good'
  },
  topic: 'ns=3;s="MyProcess"."DataFromDb";datatype=ExtensionObject;typeId=ns=3;s=DT_"DataFromDb"',
  datatype: 'ExtensionObject;typeId=ns=3;s=DT_"DataFromDb"'
}

### Test 4 - build and read with the shorter nodeId
nodeId: ns=3;s="MyProcess"."DataFromDb"

same errors as before
the second try works perfectly again

mikakaraila, 2022-08-11:
OK, node-opcua will cache most probably types. Have to ask Etienne later (on vacation at the moment).

Verschnupft, 2022-08-11:
ok, thank you :)

david114, 2022-08-22:
> OK, node-opcua will cache most probably types. Have to ask Etienne later (on vacation at the moment).

have you reached out to Etienne yet? I'm still facing this problem.

mikakaraila, 2022-09-09:
New node-opcua v2.77.0 contains now fixes.

https://github.com/node-opcua/node-opcua/releases

Verschnupft, 2022-09-09:
I'm still getting the same error :/ 
(tested it on two different Node-Red instances)

mikakaraila, 2022-09-11:
Sorry I didn´t yet update packages. I meant your could try this before I commit new package, but wait a moment.

I will publish new version with updated node-opcua.

mikakaraila, 2022-09-11:
Ok, please re-test-

Verschnupft, 2022-09-12:
still getting the error (buildNumber: 0.2.289)

Verschnupft, 2022-09-12:
I'm getting another error now. The nodeIds who are causing this are the same.
It's still only happening after the first time reading/building/...
(buildNumber: 0.2.289)
[12.09.22_Error.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9544775/12.09.22_Error.txt)

04:58:01.204Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6522
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
04:58:01.220Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6525
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
04:58:01.230Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3400
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
04:58:01.240Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3500
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
04:58:01.421Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=15889
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
04:58:01.430Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=15888
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)

mikakaraila, 2022-09-12:
@erossignon any idea what is happening?

erossignon, 2022-09-12:
@mikakaraila , I can conduct an investigation under the node-opcua support contract. This will require a live connection to the PLC. @david114 and @Verschnupft  will need to contact Sterfive for professional support.

mikakaraila, 2022-09-12:
Just one more check: did you check that namespace index is correct? I mean that nodeId is correct...
Otherwise it must be something with cache when code will populate data types.

mikakaraila, 2022-09-12:
One comment that can solve this one. Now there are 3 client nodes and I suppose cache is perhaps somehow related to session context. If you use one client node and inject msg.action with needed other parameters to same client node this could work also on first try.

mikakaraila, 2022-09-12:
OK, then it is something inside client dynamic type building/cache.

david114, 2022-09-19:
https://github.com/node-opcua/node-opcua/issues/1174#issuecomment-1250265816

mikakaraila, 2022-09-20:
As Etienne proposed try to find out problem type. I expect Tero has seen it but cannot remember what was causing this.

mikakaraila, 2022-11-06:
Please re-test with the new version that uses latest node-opcua v2.82.0

mikakaraila, 2022-11-10:
This one is still open: https://github.com/node-opcua/node-opcua/issues/1199

david114, 2022-11-11:
> Please re-test with the new version that uses latest node-opcua v2.82.0

Still occurs. But it has gotten better as I only receive three different error messages now (before 8, 9 or 10).

This is what it looks like now:
```
10:50:51.583Z :populate_data_type_manager_104:59    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6525

 ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:44:27

 ...                                                    at Generator.next (<anonymous>)

 ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)

 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)

10:50:51.655Z :populate_data_type_manager_104:59    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=15889

 ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:44:27

 ...                                                    at Generator.next (<anonymous>)

 ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)

 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)

10:50:51.665Z :populate_data_type_manager_104:59    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=15888

 ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:44:27

 ...                                                    at Generator.next (<anonymous>)

 ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)

 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)

mikakaraila, 2022-11-11:
Can you check actual extension object access levels with UaExpert?

Is there Nonatomicread?

1Tosch, 2022-11-22:
I've got the same issue here.

Writing an Extension Object with OPC UA Client:
[image]

Getting these Errors:
[image]

But the Extension Object in the server was described succesfully.

mikakaraila, 2023-01-24:
Please test with new v0.2.295 as it uses now node-opcua v2.89 that contains fixes to extension objects.

Verschnupft, 2023-01-26:
The old error messages are gone. These are the new ones (build / read / write)

```
06:51:53.298Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6522
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:51:53.308Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6525
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:51:53.316Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3400
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:51:53.324Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3500
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:51:53.457Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=15889
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
06:51:53.468Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=15888
 ...                                                    at C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\Install\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at runMicrotasks (<anonymous>)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
```

mikakaraila, 2023-01-26:
OK, @erossignon has still something to study...

david114, 2023-02-15:
Using node-opcua-client 2.90.1

Still getting error messages, but its getting a little better:

13:08:51.327Z :populate_data_type_manager_104:51     Cannot find dataType Definition ! with nodeId =ns=2;i=6525
13:08:52.894Z :populate_data_type_manager_104:51     Cannot find dataType Definition ! with nodeId =ns=2;i=15889
13:08:52.943Z :populate_data_type_manager_104:51     Cannot find dataType Definition ! with nodeId =ns=2;i=15888

mikakaraila, 2023-02-16:
Can you check that nodeId is correct for the data type?
What kind of data type you are using?

Can you share structure how does it look like in UaExpert?
Is it still this one:
[image]

mikakaraila, 2023-02-16:
NOTE: Data type nodeId namespace index is 3 not 2! Please test with correct namespace index... perhaps it will solve this.

david114, 2023-02-16:
> Can you check that nodeId is correct for the data type? Is it in correct namespace index? What kind of data type you are using? Or actually is data type nodeId like ns=3;s=MyDataType
> 
> Can you share structure how does it look like in UaExpert? Is it still this one: [image]

**Here you go:** 
[image]
[image]
[image]

Aren't those standard data types?

> NOTE: Data type nodeId namespace index is 3 not 2! Please test with correct namespace index... perhaps it will solve this.

Reading Types->DataTypes there are a lot of Data type NodeIds where Namespace Index is also 0.
Neither did I configure / program the OPC Server nor am I able to change any of the configuration.
I'm dependent on the customer here. Gonna contact him about this. Thanks so far!

mikakaraila, 2023-02-16:
Please check loaded nodesets / namespace indexes:
[image]

david114, 2023-02-16:
Those black highlighted datatypes are the ones producing error messages.
So the root of the problem seems to be the **opcfoundation.org/UA/DI** nodeset.
Maybe we can ditch it completely and see if it fixes the problem.

[image]

david114, 2023-02-17:
> Can you check DI nodeset version & date information?

Just talked to the customer (who is in charge of the PLC) and he said that he has no information about the nodesets. He's just ticking a box called "Activate OPC-Server" in Siemens TIA Software and thats it.