mikakaraila/node-red-contrib-opcua #459: Question: The memory overflow

pengju97, 2022-07-27:
hi.mikakaraila~
I've been using the point recently.
Because the PLC will often be offline due to power failure, it will often appear: [warn] [OpcUa-Client:abe42fa6f22c77b7] OpcUaClientNode: StatusCode: Bad (0x80000000)
[image]

This is the log of the problem, running for a day or so will have such a problem
I would be grateful if you could help me with this！

{"log":"24 Jul 23:24:09 - [warn] [OpcUa-Client:da7b6b58fd56fba7] OpcUaClientNode: StatusCode: Bad (0x80000000)\n","stream":"stdout","time":"2022-07-24T23:24:09.292474127Z"}
{"log":"24 Jul 23:24:09 - [warn] [OpcUa-Client:16a91f94fa9ace6b] OpcUaClientNode: StatusCode: Bad (0x80000000)\n","stream":"stdout","time":"2022-07-24T23:24:09.295447432Z"}
{"log":"24 Jul 23:24:09 - [warn] [OpcUa-Client:c4e41cc8161cc832] OpcUaClientNode: StatusCode: Bad (0x80000000)\n","stream":"stdout","time":"2022-07-24T23:24:09.29703973Z"}
{"log":"24 Jul 23:24:09 - [warn] [OpcUa-Client:bfe6aa22389c29a7] OpcUaClientNode: StatusCode: Bad (0x80000000)\n","stream":"stdout","time":"2022-07-24T23:24:09.373733425Z"}
{"log":"\n","stream":"stderr","time":"2022-07-24T23:24:15.646186977Z"}
{"log":"\u003c--- Last few GCs ---\u003e\n","stream":"stderr","time":"2022-07-24T23:24:15.646214796Z"}
{"log":"\n","stream":"stderr","time":"2022-07-24T23:24:15.646218721Z"}
{"log":"[16:0x5592dd848aa0] 54817913 ms: Scavenge (reduce) 4084.7 (4103.8) -\u003e 4084.0 (4104.3) MB, 8.7 / 0.0 ms  (average mu = 0.086, current mu = 0.012) allocation failure \n","stream":"stderr","time":"2022-07-24T23:24:15.646221386Z"}
{"log":"[16:0x5592dd848aa0] 54821044 ms: Mark-sweep (reduce) 4084.9 (4101.3) -\u003e 4083.5 (4102.1) MB, 3122.2 / 0.0 ms  (average mu = 0.063, current mu = 0.039) allocation failure scavenge might not succeed\n","stream":"stderr","time":"2022-07-24T23:24:15.646224625Z"}
{"log":"\n","stream":"stderr","time":"2022-07-24T23:24:15.646227695Z"}
{"log":"\n","stream":"stderr","time":"2022-07-24T23:24:15.64622997Z"}
{"log":"\u003c--- JS stacktrace ---\u003e\n","stream":"stderr","time":"2022-07-24T23:24:15.646232435Z"}
{"log":"\n","stream":"stderr","time":"2022-07-24T23:24:15.646234951Z"}
{"log":"FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory\n","stream":"stderr","time":"2022-07-24T23:24:15.64623724Z"}
{"log":"\n","stream":"stdout","time":"2022-07-25T00:24:35.30503576Z"}
{"log":"\u003e node-red-docker@2.1.4 start /usr/src/node-red\n","stream":"stdout","time":"2022-07-25T00:24:35.305088024Z"}
{"log":"\u003e node $NODE_OPTIONS node_modules/node-red/red.js $FLOWS \"--userDir\" \"/data\"\n","stream":"stdout","time":"2022-07-25T00:24:35.305099915Z"}
{"log":"\n","stream":"stdout","time":"2022-07-25T00:24:35.305109795Z"}
{"log":"25 Jul 00:24:35 - [info] \n","stream":"stdout","time":"2022-07-25T00:24:35.887631534Z"}
{"log":"\n","stream":"stdout","time":"2022-07-25T00:24:35.887677243Z"}
{"log":"Welcome to Node-RED\n","stream":"stdout","time":"2022-07-25T00:24:35.887687172Z"}
{"log":"===================\n","stream":"stdout","time":"2022-07-25T00:24:35.887695283Z"}
{"log":"\n","stream":"stdout","time":"2022-07-25T00:24:35.887703041Z"}

mikakaraila, 2022-07-28:
Are you inject read actions all the time even during break?

If yes then all the actions are actually buffered into the queue that will most probably cause memory overflow.

You should prevent injects or I should make some change to buffering to prevent memory overflow (like max queue size).

pengju97, 2022-08-01:
Yes, I can't tell when the equipment is down，I can only keep the OPCUA client running to maintain data continuity，Can you give me some ways to optimize？
Thank you for your reply~~~

mikakaraila, 2022-08-01:
You could check client node status and disable inject read or write actions as the connection is broken.
But I will study more if I could add some limit to queue.

mikakaraila, 2022-09-05:
If no news I will close this. You can reopen in case of some news.

Is there running other nodes than OPC UA?

Docker can have some limitations, I haven´t seen these messages anywhere else:
ms: Scavenge (reduce) 4084.7 (4103.8) -\u003e 4084.0 (4104.3) MB, 8.7 / 0.0 ms (average mu = 0.086, current mu = 0.012) allocation failure \n","stream":"stderr","time":"2022-07-24T23:24:15.646221386Z"}
{"log":"[16:0x5592dd848aa0] 54821044 ms: Mark-sweep (reduce) 4084.9 (4101.3) -\u003e 4083.5 (4102.1) MB, 3122.2 / 0.0 ms (average mu = 0.063, current mu = 0.039) allocation failure scavenge might not succeed\n","stream":"stderr","time":"2022-07-24T23:24:15.646224625Z"}

Could be related somehow to invalid url if remote computer is not accessible:
https://stackoverflow.com/questions/32271450/how-to-avoid-fast-memory-increase-during-scavenge-gc

closed 2022-09-05 by mikakaraila