mikakaraila/node-red-contrib-opcua #460: Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3400

jadevicpavel, 2022-07-27:
Hello,
I have some trubles with OPC UA browser. I created script that loads all objects of my custom unit, basically I input my custom path and broweser returns an object payload with all children of my custom object. I tried to writed everything dynamically so I could reuse my script on other projects. When I did so and changed ip, OPC UA broweser started to trow new error (like 10000+ times) and it becomes much slower. 

Error names: 
populate_data_type_manager_104:38    Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3400
populate_data_type_manager_104:167   err= Error

I include pictures of both old working project and new error project. Even thoe it trows multiple errors OPC UA broweser returns payload with all children but as I said before its much slower and problem is I upload to to web and speed is esential for me. We are talking about 30 sec+ per unit and there are over 80 units.
another problem is that old project was developed in _Tiaportal16_ for PLC cpu 1518 and new poject was developed in _Tiaportal 17_. I havent noticed any changes in properties but still I think its worth mentioning.

Versions:
node-red = 2.2.2
opc-ua = 0.2.260

* I've tried to upgrade to newer versions and problem remained. 
[image]

mikakaraila, 2022-07-28:
There has been changes on node-opcua: https://github.com/node-opcua/node-opcua/issues/1174

We have to wait until those are solved.

jadevicpavel, 2022-07-28:
@mikakaraila 
Thank you for fast replay.
I did some testing on my own and come to some interesting conclusion, I think main problem lies in Tiaportal verision because project developed in _Tiaportal 16_ works well after upgrading this same project to _Tiaportal 17_ it starts to throw errors mantion both by me and issue you refrenced. 

I work in company that has direct connection to Siemens and

jadevicpavel, 2022-07-28:
@mikakaraila
1) I accidentally closed this issue, shloud I open new one or can we contionue in this closed ?
2) Thank you for fast replay.
I did some testing on my own and come to some interesting conclusion, I think main problem lies in Tiaportal verision because project developed in Tiaportal 16 works well after upgrading this same project to Tiaportal 17 it starts to throw errors mantion both by me and issue you refrenced.

I work in company that has direct connection to Siemens and

jadevicpavel, 2022-07-28:
@mikakaraila
Thank you for fast replay.
I did some testing on my own and come to interesting conclusion, I think main problem lies in Tiaportal verision because project developed in Tiaportal 16 works well after upgrading this same project to Tiaportal 17 it starts to throw errors mantion both by me and issue you refrenced.

I work in company that has direct connection to Siemens support and we can provide direct connection between you and them to help resolve this problem directly with source.

mikakaraila, 2022-08-02:
I expect @erossignon could help also. Have you checked namespace index is correct?

jadevicpavel, 2022-08-02:
There were no namaspace changes as I understand because we upgraded working project to _tiaportal17_ and unleass upgrade it self changed namaespace, there should not be problem. In both cases path ns=3;i=3400 leads to SimaticStructures.
[image]

mikakaraila, 2022-08-10:
Please test, I updated dependency node-opcua v2.73.0 that contains fixes related to this.

mikakaraila, 2022-09-09:
New node-opcua v2.77.0 contains fixes to entension object.

https://github.com/node-opcua/node-opcua/releases

jadevicpavel, 2022-09-12:
I've updated to last version and tried to run script. Unfortunatelly for me it didn't resolve the problem. Except error with nodeId =ns3;i=3400 there are 5 new nodeId errors:
nodeId =ns=2;i=6522
nodeId =ns=2;i=6525
nodeId =ns=3;i=3500
nodeId =ns=2;i=15888
nodeId =ns=2;i=15889

[image]

And same problem remaines OPC broweser can't find schemas in SimuticStructures even thoe they are there but isntead of error message, now there is notation fix me!

[image]

@mikakaraila

mikakaraila, 2022-09-12:
Ok, perhaps @erossignon  can find out reason why data type manager cannot handle those.

There is support for v1.03 and v1.04 but older ones can be still problem or Siemens has a bit different implemention.

cdac2019, 2022-09-21:
@mikakaraila  @jadevicpavel @erossignon  currently I ave updat
[image]
ed latest pkg showing such type of issues

mikakaraila, 2022-09-21:
This is inside node-opcua, there is some special Siemens datatype that cannot be converted.

If you can find it report it to this issue: https://github.com/node-opcua/node-opcua/issues/1174
For some reason it seems to be closed??

mikakaraila, 2025-06-27:
I expect this can be closed

closed 2025-06-27 by mikakaraila