mikakaraila/node-red-contrib-opcua #461: Question: How to call method sequently?

Huang2Ning, 2022-08-01:
Hello,

I want to create a flow that calls the OPCUA method sequently like this
[image]

But the result shows that the first method has been executed twice, while the second method hasn't been called. We have carefully verified the nodeId of the owner and method. And when we have a larger flow that calls several methods one after another, it seems that the server is always receiving the same method again and again. 
[image]

I'm afraid that I have been using the node wrongly, could you please help me to reform the flow? Thank you in advance.

mikakaraila, 2022-08-02:
You have to clear msg before or after delay, otherwise it contains first method call parameters => calls again it.
Call two methods:
[image]

Clear msg:
[image]

mikakaraila, 2023-02-28:
Closed as solved long time ago.

closed 2023-02-28 by mikakaraila