mikakaraila/node-red-contrib-opcua #462: Connect Siemens OPC UA Client to Node-RED OPC UA Server

AndreSchm, 2022-08-02:
We have a Node-RED OCP UA Server running on a local Windows 10 PC using node-red-contrib-opcua v0.2.283. Now we want to connect a [Siemens S7 OPC UA Client](https://support.industry.siemens.com/cs/document/109762770/s7-user-block-for-the-opc-ua-client-of-a-simatic-s7-1500?dti=0&lc=en-US) (based on PLCopen) to the server in a first step without any security policy and mode using Siemens TIA Portal v17. We are able to scan and find the server and its adress space in the local network using the PCs IPv4 adress. But when we send a client-side Connect Control we get a very general error code "16#8005_000". Connections from Node-RED and UaExpert clients work fine using the systems hostname or IPv4 adress.

1. Is it possible to configure the servers adress? Changing the endpoint in Node-REDs Global Confoguration Nodes did not have any effect.
2. Did someone realise a S7 connection to a Node-RED Server successfully or is familar with the described connection issue?

Thank you very much in advance.

mikakaraila, 2022-08-02:
Could it be firewall that blocks port?
It is not node-red / node-opcua specific as the same error seems to come with KepServer:
https://support.industry.siemens.com/forum/WW/en/posts/opc-ua-connection-error-16-8081-0000/272592

AndreSchm, 2022-08-03:
We could establish a connection to a software PLC using Siemens PLCSIM today successfully. I will try the same configuration on the hardware PLC later. Nevertheless is it possible to set a fixed IP to the Node-RED Server?

AndreSchm, 2022-08-03:
It was indeed an IT issue. After reconfiguration of the used network adapter and a memory reset in the Siemens PLC we are able to establish the connection to the Node-RED Server, read and write values.

closed 2022-08-03 by AndreSchm