mikakaraila/node-red-contrib-opcua #463: Error in Version "2.70.3" of node-opcua-client-dynamic-extension-object => ExtensionObject can not resolved

ThisUsernameIsValid, 2022-08-05:
Hello mikakaraila,

first thanks to the great library!

I just want to send a bug-report to https://github.com/node-opcua but I saw the topic was solved on 24. June.
Can you check if version 2.72.2 can be used with your package, and publish a version with the new dependencies?

[image]

Here is my not sent bug-report:

**Current behavior**

When in function "convertDataTypeDefinitionToStructureTypeSchema"  "definition.fields" is null or undefined then the for loop can't iterate and brings an exception.

**Describe the bug**
Log, after the error occures:
09:48:42.806Z :populate_data_type_manager_104:38    Error TypeError: definition.fields is not iterable
 ...                                                    at C:\...\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\convert_data_type_definition_to_structuretype_schema.js:268:37
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at C:\...\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\convert_data_type_definition_to_structuretype_schema.js:8:71
 ...                                                    at new Promise (<anonymous>)
 ...                                                    at __awaiter (C:\...\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\convert_data_type_definition_to_structuretype_schema.js:4:12)
 ...                                                    at convertDataTypeDefinitionToStructureTypeSchema (C:\...\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\convert_data_type_definition_to_structuretype_schema.js:247:12)
 ...                                                    at C:\...\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\convert_data_type_definition_to_structuretype_schema.js:159:40
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\...\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\convert_data_type_definition_to_structuretype_schema.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
09:48:42.810Z :resolve_dynamic_extension_obje:75    err TypeError: Constructor is not a constructor
 ...                                                    at C:\...\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\resolve_dynamic_extension_object.js:57:28
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\...\.node-red\node_modules\node-opcua-client-dynamic-extension-object\dist\resolve_dynamic_extension_object.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
09:48:42.814Z :resolve_dynamic_extension_obje:76    opaqueStructure =  ns=11;i=5011
09:48:42.817Z :resolve_dynamic_extension_obje:77    opaqueStructure =  0x000000000000000000000000000000000000000010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
09:48:42.818Z :resolve_dynamic_extension_obje:78    00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 10 00 00 00 00 00 00 00 00 00 00 00    ................................
 ...                                                00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................................
 ...                                                00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................................
 ...                                                00000060: 00 00 00  

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

definition.toString() of type with "null[]"-Field:
{ /*StructureDefinition*/
 ...                                                 defaultEncodingId             /* NodeId              */: ns=11;i=5009
 ...                                                 baseDataType                  /* NodeId              */: Structure (ns=0;i=22)
 ...                                                 structureType                 /* StructureType       */: 0 /*(Structure)*/
 ...                                                 fields                        /* StructureField   [] */: null []
 ...                                                };

Structor of type with "null[]"-Field:
[image]

**Expected behavior**

After I have inserted following check, it works for me: 
if (definition.fields !== undefined && definition.fields !== null) {
     for (const fieldD of definition.fields) {
     ...
     }
}

**Screenshots**

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( ) I have installed node-opcua as a package ( using npm install )
- (x) I am using an application that uses node-opcua

    - (x) node-red
    - ( ) other : please specify

- Device: Windows-PC
- OS version:

    - (x) Windows : version : 10
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: Siemens-PLC CPU1515
    - Version: V2.8, TIA 16.3, PLC SIM Advanced V4
    - Manufacturer/Software vendor: Siemens

- **node-opcua version:** :
- node-red-contrib-opcua 0.2.284
Package of invalid-function:
{
    "name": "node-opcua-client-dynamic-extension-object",
    "version": "2.70.3",
    "description": "pure nodejs OPCUA SDK - module client-dynamic-extension-object",
    "main": "./dist/index.js",
    "types": "./dist/index.d.ts",
    "scripts": {
        "build": "tsc -b",
        "clean": "npx rimraf dist *.tsbuildinfo",
        "test": "echo no test"
    },
    "dependencies": {
        "chalk": "4.1.2",
        "node-opcua-assert": "2.66.0",
        "node-opcua-binary-stream": "2.69.0",
        "node-opcua-constants": "2.70.0",
        "node-opcua-data-model": "2.70.0",
        "node-opcua-data-value": "2.70.0",
        "node-opcua-debug": "2.69.0",
        "node-opcua-extension-object": "2.70.0",
        "node-opcua-factory": "2.70.0",
        "node-opcua-nodeid": "2.70.0",
        "node-opcua-pseudo-session": "2.70.3",
        "node-opcua-schemas": "2.70.3",
        "node-opcua-service-browse": "2.70.3",
        "node-opcua-service-translate-browse-path": "2.70.3",
        "node-opcua-status-code": "2.70.0",
        "node-opcua-types": "2.70.3",
        "node-opcua-variant": "2.70.0"
    },
    "author": "Etienne Rossignon",
    "license": "MIT",
    "repository": {
        "type": "git",
        "url": "git://github.com/node-opcua/node-opcua.git"
    },
    "keywords": [
        "OPCUA",
        "opcua",
        "m2m",
        "iot",
        "opc ua",
        "internet of things"
    ],
    "homepage": "http://node-opcua.github.io/",
    "gitHead": "08b5465af631fd010bc48843c951b2e556270de4"
}

mikakaraila, 2022-08-05:
Ok, I just updated our product to use v2.72.2

I will try to take a look and fix other things at the same time (but not today). First week at work after vacation.

ThisUsernameIsValid, 2022-08-05:
This sounds good!

Thanks for the quick respond!

mikakaraila, 2022-08-10:
Please test, I updated dependency node-opcua v2.73.0 that contains fixes related to this.

mikakaraila, 2022-09-09:
I think I should update to node-opcua v2.77.0 it contains more fixes for extension object handling.

https://github.com/node-opcua/node-opcua/releases

mikakaraila, 2022-10-10:
Latest version of node-opcua is v2.79.1 should be tested, can you test it?

Another note from Siemens support discussion:
"If you cannot get an access to the Specialised DBs of the PID controller then a workaround could be to copy the used Data to an normal Datablock in order to get normal access over OPC Server. so you can create an Interface DB for OPC communication."

mikakaraila, 2022-11-06:
Please re-test with the new version that uses latest node-opcua v2.82.0

mikakaraila, 2022-11-10:
This one seems to be still open: https://github.com/node-opcua/node-opcua/issues/1199

mikakaraila, 2023-01-24:
Please test with new v0.2.295 as it uses now node-opcua v2.89 that contains fixes to extension objects.

mikakaraila, 2023-07-04:
If no testing... what can I do? Closed!

closed 2023-07-04 by mikakaraila