mikakaraila/node-red-contrib-opcua #464: OpcUa Method - connection-error & suggestion for new Method-Node-Structure

ThisUsernameIsValid, 2022-08-05:
Hello mikakaraila,

I've some other issue... 
Can you please check the "Method-Node"? I'had here the problem, that I can't build up the connection - the "Client-Node" works however with the same settings. 
After looking in the code I found some differences between "Client-Node"-connection-setup and "Method-Node"-connection-setup. I copied some lines, than the connection works - but I don't know if it works with security-settings (e.g. certificates) - there are still some differences...

I think the following structure makes it easier for you to maintain:
- convert the Method-Node into an only-configuration-node, like OpcUa-Item. 
- add in Client-Node an action, to call functions
- implement a command-interpreter in Client-Node, that one can connect this node or disconnect this from outside

Please keep the option to control the number of in/outpus of Method-Node - via code its for me OK ;-)

mikakaraila, 2022-08-08:
OK, I will take a look  to connect issue later, but my tests have passed.

Good proposals for a new features, thanks.

New v2.72.2 will allow client to disconnect & connect, earlier it was not possible.

mikakaraila, 2022-09-01:
I added method call into the client node:

[image]

ThisUsernameIsValid, 2022-09-01:
**Thank you for these very great news!** 
I already solved this laborious by implementing it manually with write/read-commands, but with this optimization it would be much cleaner!

Can you publish the code behind these shown flows as example?

mikakaraila, 2022-09-01:
I implemented also connect and disconnect earlier.

I will add example flows tomorrow. Time to sleep next...

mikakaraila, 2022-09-04:
Forgot, but here:
[METHOD.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9484096/METHOD.txt)

mikakaraila, 2022-09-05:
Open if something is still missing or unclear.

closed 2022-09-05 by mikakaraila