mikakaraila/node-red-contrib-opcua #469: [Feature Request] Make client endpoint dynamically configurable

pascalporedda, 2022-08-23:
Today I ran into the problem that I need the OPC client's node endpoint to be configurable, either via selecting an env variable, or setting the endpoint to an incoming messages property. Like, it is possible with many other nodes in NodeRED.

Would it be possible for you to add this feature?

mikakaraila, 2022-08-23:
It is already existing, look OPCUA-TEST-NODES.json

I will extend actions to support disconnect & connect. Current implementation uses action "reconnect".

pascalporedda, 2022-08-24:
Yeah I can verify that this is working as expected. Though I would still prefer being able to select end endpoint url from an ENV variable for example, because now I have to set the UaEndpoint everytime I call a method as well for example. Having the option to refer to an env variable inside the UA config node would make things easier.

mikakaraila, 2022-08-24:
Did you look older example? It will use msg that contains endpoint information...

pascalporedda, 2022-08-24:
Yes I've seen that, using a msg works fine but I would like to have this:

[image]

Instead of in a change node that sends a message to the OPC client rather here: 

[image]

mikakaraila, 2022-08-24:
Argh, can you change environment variable value during runtime?
Or use function node to build value and then inject msg.

mikakaraila, 2022-08-24:
Do you have example where this feature has been used?

pascalporedda, 2022-08-25:
Perfect, thank you. Yeah the Inject Node covers this feature.

tedkim13, 2022-08-25:
Hi all, I believe it's already possible to use an environment variable for configuring the OpcUa-Endpoint node, we do something similar for our use case:

[image]

One thing to note is that the environment variable must be the full endpoint. Unfortunately cannot substitute just a part of the property. e.g. `opc.tcp://{ENDPOINT}:48030`
[image]

Hope this helps.

pascalporedda, 2022-08-26:
Hi @tedkim13 this looks promising, I'll try this 👍🏻 and leave you guys feedback once I know if this works or not.

pascalporedda, 2022-08-26:
Works perfectly fine. Thank you!

closed 2022-08-26 by pascalporedda

offrok, 2023-02-27:
@pascalporedda could you please explain how you set the environment variable at runtime in this context? I thought you could only set them at startup

afagundesm, 2023-09-11:
Hi, using the example OPCUA-TEST-NODES.json (tab reconnect) I am able to use the action "reconnect" in a function node but then when I try to use the action "browse" it will not browse on the new endpoint address space (after reconnecting). Instead, it will try to browse on the endpoint initially configured for the OPC UA Client node. There is a way of giving the msg.OpcUaEndpoint on the action "browse" like we do in "reconnect"?