mikakaraila/node-red-contrib-opcua #474: is there a way to disable warnings

JanEggers, 2022-09-01:
i get a bunch of warnings when setting values.

Is there any way to disable them?

i am spammed by them and the dev console is basicly useless.

[image]

Also the buildin filter does not work / show the nodes so i can deselect them. 

[image]

mikakaraila, 2022-09-01:
Change node-red settings.js logLevel to error or fatal:

// Configure the logging output
    logging: {
        // Only console logging is currently supported
        console: {
            // Level of logging to be recorded. Options are:
            // fatal - only those errors which make the application unusable should be recorded
            // error - record errors which are deemed fatal for a particular request + fatal errors
            // warn - record problems which are non fatal + errors + fatal errors
            // info - record information about the general running of the application + warn + error + fatal errors
            // debug - record information which is more verbose than info + info + warn + error + fatal errors
            // trace - record very detailed logging + debug + info + warn + error + fatal errors
            // off - turn off all logging (doesn't affect metrics or audit)
            level: "debug",
            // Whether or not to include metric events in the log output
            metrics: false,
            // Whether or not to include audit events in the log output
            audit: false
        }
    },

JanEggers, 2022-09-01:
seems like you commented out the seetings

https://github.com/mikakaraila/node-red-contrib-opcua/blob/95dbc2aab1ddd4f17cb531ee7ec97dda77988288/opcua/101-opcuaitem.js#L37-L39

JanEggers, 2022-09-01:
and in general i want to se info messages just not this one

mikakaraila, 2022-09-01:
These changes were wanted by another person to reduce messages.

Comment out lines 68 and 75 to get rid of these:
[image]

mikakaraila, 2022-09-01:
I will publish new version later today...

closed 2022-09-02 by JanEggers