mikakaraila/node-red-contrib-opcua #476: Can't connect to the opcua server while using [SecurityMode:None] and [SecurityPolicy:None]

Jireyo, 2022-09-02:
the console displayed:
Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:[1] securityPolicy:["http://opcfoundation.org/UA/SecurityPolicy#None"]
Case B: UserName & password does not match to server (needed by Sign): undefined undefined

But the securitymode and securitypolicy has been set to none,and this question only happened in recent version.
Now i'm using v0. 2.223,it connected.

mikakaraila, 2022-09-02:
If you use Sign then username & password must be provided.
Case B row shows: undefined / undefined

Jireyo, 2022-09-02:
Here is my endpoint-setting,i use anonymous mode and save it.when i opened the endpoint-setting again,the choose (anomymous) disappeared.
[image]
And through my current test,the endpoint can connect server at version 0.2.281,but at latest version,the issue happend.

> If you use Sign then username & password must be provided. Case B row shows: undefined / undefined

mikakaraila, 2022-09-02:
OK some bug in the UI/settings... I will check later.

mikakaraila, 2022-09-04:
Fixed to v0.2.288

HL4096, 2022-09-09:
I'm having the same problem with the latest version 0.2.288
On another instance I have version 0.2.284 and this runs fine

Jireyo, 2022-09-09:
> 

same question,the UI has been fixed,but setting still have some problem.

mikakaraila, 2022-09-09:
Can you give more details when this happens. Instructions to reproduce?
I tested this couple of times and different ways, bit I didn´t catch how it will happen.

HL4096, 2022-09-09:
Here are two instances with same flow connecting to same OPCUA and MQTT
First one is 0.2.284 working fine, 'active subscribed'
Second one is 0.2.288 "connecting"
Same settings for each, I have tried uninstalling and trying an older version but cant get it to work
[image]

[image]
[image]
[image]

HL4096, 2022-09-09:
I have tried with/without anonymous, the instance that works does not have it selected

mikakaraila, 2022-09-09:
Default value is true, so there is something that makes it false.

HL4096, 2022-09-09:
its being declined because it has no password, see log

[image]

mikakaraila, 2022-09-11:
OK, I will take second look... just a moment

mikakaraila, 2022-09-11:
Please re-test.

HL4096, 2022-09-11:
After updating to 2.2.289 and restarting, I am getting private key error
[image]
[image]
[image]

mikakaraila, 2022-09-12:
I tested this before publishing and now again. Cannot reproduce this.

Any hints or instructions how to reproduce this?

mikakaraila, 2022-09-12:
Please check all client nodes if there is still one with wrong settings...

HL4096, 2022-09-12:
Saw this in the log so I manually dl the file and restarted.  all is good now!
thanks for your help
[image]

P1Rebo, 2022-09-12:
Same here, checked both "anonymous = true" or left empty. If I try to connect to the OPC UA Server with a standard client it works fine. Worked also with previous versions. Environment is Node Red within docker environment, newest Node Red image. No connection is possible. The solution described by @HL4096 doesn't obviously work for me.

Rollback is not possible. I wasn't expecting such kind of changes in a minor version... (anyhow no backup / sandbox test no mercy, I will do this in future for sure ...).

If there is anything I can test, please tell. Thanks for issue fix.

mikakaraila, 2022-09-13:
I need at least something like console output to see what is happening... as I wrote I cannot reproduce this with any of my tests.

Most probably root cause is missing openssl.

Jireyo, 2022-09-13:
Updated to version 0.2.289 and it works well. Thanks for your fix.

P1Rebo, 2022-09-13:
I had a look in the log.

`13 Sep 09:00:29 - [error] [OpcUa-Client:dd5d9ccc.94ce7] Error: EACCES: permission denied, mkdir '/root/.config/node-red-opcua-nodejs/PKI'`

I tried to start Node Red in docker with privileged mode but it didn't fix the problem.

mikakaraila, 2022-09-13:
OK, I think I will close this now. Most probably these same errors will popup again later...

closed 2022-09-13 by mikakaraila

youngkzy, 2022-09-29:
@mikakaraila ,
  I downloaded openssl-1.0.2o-x64_86-win64.exe for Windows 10 64 bit VM.  I log in as machine root administrator and ran Node-RED with this file in the Node-RED directory as per the folder was opened by the system.  I unzipped the file to this folder.  This solved the issue ONLY for machine root admin.  Node-RED runs from SYSTEM since it is started via NSSM.  However, If I start Node-RED under another user, you now need to have it in the proper folder as per the user.  
Error:  cannot find file C:\Users\<userlogin>\AppData\Local\Programs\openssl\openssl.exe
openssl seems to be missing and need to be installed

I'm assuming the install was not proper.  How is this recommended for proper install of openssl?
Thanks

mikakaraila, 2022-09-30:
node-opcua-pki package takes care of openssl, it will require internet connection to make installation.
If there is no internet connection it should be installed manually if OS doesn´t contain openssl.
By design PKI folder used to store certificates (trusted/rejected) is user specific (it is very compilated to support multiple environments).