mikakaraila/node-red-contrib-opcua #477: Errors don't trigger Catch-Nodes

Verschnupft, 2022-09-09:
Hello,

is there a specific reason, why the client-nodes are throwing errors by using _node.error('text')_ instead of _node.error('text', msg_)? 
We would be able to directly work with the errors since catch-nodes are only triggerd by the second option.  

> From the [documentation 114](https://nodered.org/docs/creating-nodes/node-js#handling-errors):
> If the error is one that a user of the node may want to handle for themselves, the function should be called with the original message (or an empty message if this is an Input node) as the second argument:
> node.error(“hit an error”, msg);
> This will trigger any Catch nodes present on the same tab.
> So calling node.error() with one argument will not trigger a Catch node.

(cp. https://discourse.nodered.org/t/node-error-not-caught-in-catch-node/1301/2).

mikakaraila, 2022-09-09:
Implementation is originated 2015 or something...  have to fix when I will have time.

mikakaraila, 2023-01-24:
Please test with v0.2.295

I fixed selected placed that I found would be meaningful to catch. Other errors will still come as earlier.
If you found that something should be added "catchable" please inform.

Verschnupft, 2023-01-25:
thanks c:

closed 2023-01-25 by Verschnupft