mikakaraila/node-red-contrib-opcua #479: Invalid message header detected

TobiasKorn, 2022-09-14:
Hello community,

after updating the Firmware of our OpenController (V2.8->V2.9) secure communication between NodeRed and the OPC Server I get following error at the client node:

**invalid endpoint Error: The connection may have been rejected by server, Err = (Invalid message header detected)**

If i use no security policy, security mode no, and anonymous login I get connected. I also don't have certificate problems, as far as i can check it. And with the same settings in NodeRed it worked, before i made the Firmware update.

Can somebody help me?

Greetings
Tobias

mikakaraila, 2022-09-14:
Please contact commercial support from @erossignon  as this seems to be protocol level case.

Or rebrowse endpoint with UaExpert and check that you have same opc.tcp://hostname:port/resourcePath/ at the client node.

TobiasKorn, 2022-09-14:
Thank you for your quick reply. With UaExpert i have no problems to access the server (with security policy Basic256Sha256, Sign & Encrypt with username and password).

I do not use a resourcePath. In UaExpert and NodeRed my EndpointUrl is both opc.tcp://hostname:port
Can this make a difference to use a resource path? What resourcePath can I use there?

mikakaraila, 2022-09-14:
Just same as with UaExpert if no path then you can leave endpoint url as it is. 
There is another issue with the same Basci256Sha256, I will check it....

TobiasKorn, 2022-09-16:
I also tested it with security police Basic128Rsa15 and Basic256, but they also don't work.

By the way: should I write to @erossignon private e-mail with he provides in his account or is there another way to reach him? I am asking because he didn't reply to my email i wrote him two days ago.

mikakaraila, 2022-09-16:
Or use Wireshark to see what is happening on network. There is filter for opcua.

One more time it is good to check node-red console as there has been sometimes messages telling actual error.

TobiasKorn, 2022-09-16:
NodeRed console puts out:

**TIMESTAMP: client_secure_channel_layer    :206 requestData not found for requestId = 1 try with 2**

[image]

Wireshark see as follows:

[image]

mikakaraila, 2022-09-16:
Test with this environment variable:
set NODEOPCUADEBUG=CLIENT{REQUEST-RESPONSE-TRACE-STATS-PERF}
Then start node-red, it should give more details about client messages...

TobiasKorn, 2022-09-19:
I am not sure how i can apply what you said, to get more diagnostic information. I set up an environment variable in the flow tab, where i use the OPC UA Client node

[image]

But when i read the value of the environment variable, i just get an empty string.

mikakaraila, 2022-09-19:
It is on the command line level environment. Then loaded node-opcua will produce new messages to the console.

TobiasKorn, 2022-09-19:
Indeed i get a lot more information. I pasted a couple of lines, I got, because i couldn't directly recognize a pattern, which would be repeating every few seconds.

[NodeRedConsoleLog.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9597022/NodeRedConsoleLog.txt)

mikakaraila, 2022-09-20:
Did you notice this line:
 .... TRUNCATED ..... (NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE=25

Add more lines to see the whole message:
set NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE=100

Same way as the other environment variable and start node-red again to get more information.

TobiasKorn, 2022-09-20:
I had a conversation with @erossignon and we create together the attached logfile. Maybe this can also help you to find the error, even though @erossignon told me, that probably he needs to do changes in his software. So the problem might be out of your scope.

[log.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9605947/log.txt)

The log file was created by running the following JavaScript (change data extension .txt -> .js)

[clienttest.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9605963/clienttest.txt)

mikakaraila, 2022-09-20:
Seems to use wrong TokenId for some reason... we have to wait until Etienne has solved this one.

TobiasKorn, 2022-09-20:
Maybe he will, maybe not. For our application we cannot justify to pay the price he wants to have for error investigation. So he said he will maybe do it, when there is nothing else important on his schedule.

Jaxom99, 2022-11-02:
I have the same situation with an Ignition server... I will try to look into your debug procedures and provide my logs (from a Linux host).

mikakaraila, 2022-11-03:
It would be good to report findings to @erossignon

TobiasKorn, 2022-11-08:
I just want to let you know, that one of the last updates has fixed the error

closed 2022-11-10 by mikakaraila