mikakaraila/node-red-contrib-opcua #480: Secure connection between SIMATIC S7-1500-OPCUA-Server and Node-red-OPCUA-Client -> Client´s server certificate verification has failed

fespGIT, 2022-09-14:
Hello,

i got two virtual machines running on my Host. 
Virtual Machine 1 (Windows) is simulating a SIMATIC S7-1500 from Siemens. The S7-1500 is running an OPC UA Server. 
Virtual Machine 2 (debian) is running node-red. With the node OPC UA-Client i want to read a variable from the S7-1500.
Everything works without using Security-Settings and Certificates.

I created a CA-Certificate and with this a normal Certificate on my S7-1500.
I exported them as P12 from VM1 to VM2. In VM2 i used 
openssl pkcs12 -in path.p12 -out newfile.crt.pem -clcerts -nokeys
to get pem-files.
Then i copied the CA-Certificate in
/home/fesp/.config/node-red-opcua-nodejs/PKI/issuers/certs/
and the Certificate in
/home/fesp/.config/node-red-opcua-nodejs/PKI/trusted/certs/.

The Client-node is configured as in the pictures. The Endpoint-IP with port should be correct as it worked without Certificates and Security-Settings.
SecurityPolicy and SecurityMode is the same i gave my S7-1500. I created a user in my S7-1500 and wrote the credentials inside the client-node.

I already tried to use own Certificates for the client which i created with XCA on VM2.
I stored them in
/home/fesp/.config/node-red-opcua-nodejs/PKI/own/certs/
and the keys in
/home/fesp/.config/node-red-opcua-nodejs/PKI/own/private/.
I wrote the paths into the client-node.
I also exported them to the S7-1500 and registered them as trusted.

Same output.

After starting Node-Red I get this output from my terminal:

[image]
[image]
[image]

What do i need to do, that the client will trust the Server-Certificate?
And why i am getting the warnings Case A and Case B?
The Endpoint (S7-1500) has the setting sign&encrypt.
The username and password should also match to server.

I hope you can help me.

mikakaraila, 2022-09-14:
Double check that you have openssl installed. It seems to be missing if there is no internet connection from vm.

mikakaraila, 2022-09-15:
Your folder structure is OK, my environment is different as I use PKI Folder and OS is Windows.
Check permissions! 
Node-red should have access to all folders and also write access if client stores server certificate to trusted folder.

fespGIT, 2022-09-15:
I checked Permissions. PKI folder can be used in every way by every user now.
Still no secure connection possible.

Thank you for your help by the way :-)

mikakaraila, 2022-09-15:
You can turn on debug level trace, look settings.js

mikakaraila, 2022-09-15:
Check also server if it will need to trust client certificate.

fespGIT, 2022-09-16:
While trying to connect the server to UAExpert-Client, i recognized that client needs the certificate revocation lists.
Adding the .crl of ca and .crl of certificat (server) to issued/crl an trusted/crl (NodeRed-Client) solved the problem.
So a secure connection to nodered is now also possible.

thanks and greetings

mikakaraila, 2022-09-17:
Excellent, could you still add some example here. Most probably this will come up with other people.

fespGIT, 2022-09-17:
To connect the SIMATIC S7-1500-OPCUA-Server with UAExpert-Client securely, you can use the Document below as a guidline.
https://support.industry.siemens.com/cs/document/109769068/using-certificates-with-tia-portal?dti=0&dl=en&lc=de-WW

For secure connection between SIMATIC S7-1500-OPCUA-Server and Node-RED-OPCUA-Client (on debian11):
-do the same as you did for secure connection to UAExpert-Client (in your TIA-Portal-Project)
-export CA from TIA-Portal as .der (certificate) an .crl (certificate revocation list) and store them under following paths in debian:
/home/"yourusername"/.config/node-red-opcua-nodejs/PKI/issuers/certs
/home/"yourusername"/.config/node-red-opcua-nodejs/PKI/issuers/crl
- export servers certificate as .der and .crl as well and store them under following paths in debian:
/home/"yourusername"/.config/node-red-opcua-nodejs/PKI/trusted/certs
/home/"yourusername"/.config/node-red-opcua-nodejs/PKI/trusted/crl
-under follwing paths were clients certificat and private key, i used them but i think you can create your own and copy them to the paths to use own certificates for client
/home/"yourusername"/.config/node-red-opcua-nodejs/PKI/own/certs/client_certificate.pem
/home/"yourusername"/.config/node-red-opcua-nodejs/PKI/own/private/private_key.pem
-Import the file client_certificate.pem as a trusted certificate into your TIA-Portal-Project
in node OpcUa-Client:
-set security-settings as you defined in TIA-Portal for Endpoint (Endpoint opc.tcp://serversIPadress:port as you can see in your TIA-Portal-Project) , i used  Basic256Sha256, Sign&Encrypt, "use credentials"
-certificate: local certificate file, ....
-give paths of your local client-certificate and key and PKI folder as well

secure connection should work now

mikakaraila, 2022-09-17:
Thanks a lot. I don't have any S7-1500 nor TIA.

closed 2023-06-28 by mikakaraila