mikakaraila/node-red-contrib-opcua #482: how to use both sampling interval and publish interval on subscribe action

vdwpsmt, 2022-09-27:
Hi, 

I'm still struggling on the mixed usage of the 2 types of intervals for monitored data items: 
- samplinginterval
- publishinterval

For my application I want to use both parameters but I'm not sure whether both are implemented in the code.
To have a clear understanding, this figure shows what is meant by those 2 terms in the OPC-UA context:

[image]

in the documentation for the opcua client node of node-red-contrib-opcua I found this:

[image]
So the assumption I made was:
- msg.payload = samplinginterval
- msg.interval = publishinterval

So my question is: is this assumption correct?  Is it possible to set the sampling interval? What is the default value for the intervals?

mikakaraila, 2022-09-27:
Snippets from the implementation:
```JavaScript
     var interval = convertAndCheckInterval(msg.payload);
        verbose_log(msg.topic + " samplingInterval " + interval);
...
  const monitoringParameters = {
          samplingInterval: interval,
          queueSize: 1,
          discardOldest: true
        };
```

And with events it will use 100ms as default:
 requestedPublishingInterval: timeMilliseconds || 100,

For publish I didn´t find anything immediately... sorry.

mikakaraila, 2022-09-27:
Found publishInterval these defaults:
```JavaScript
module.exports.getSubscriptionParameters = function (timeMilliseconds) {
    return {
        requestedPublishingInterval: timeMilliseconds || 100,
        requestedLifetimeCount: 30,
        requestedMaxKeepAliveCount: 3,
        maxNotificationsPerPublish: 10,
        publishingEnabled: true,
        priority: 10
    }
};
```

vdwpsmt, 2022-09-28:
Could it be possible that in the case of a single subscription (action = subscribe and  topic = nodeId), msg.payload and msg.interval have the same function (publishinterval)?  not sure if I'm looking at the right place in the code but this is what I found: 
```javascript
async function subscribe_monitoredItem(subscription, msg) {
.
.
.

      if (!monitoredItem) {
        verbose_log("Msg " + stringify(msg));
        // var interval = 100; // Set as default if no payload
        var queueSize = 10;
        var interval = opcuaBasics.calc_milliseconds_by_time_and_unit(node.time, node.timeUnit); // Use value given at client node
        // Interval from the payload (old existing feature still supported), but do not accept timestamp, it is too big
        if (msg.payload && parseInt(msg.payload) > 100 && parseInt(msg.payload) < 1608935031227) {
          interval = convertAndCheckInterval(msg.payload);
        }
        if (msg.interval && parseInt(msg.interval) > 100) {
          interval = convertAndCheckInterval(msg.interval);
        }
        if (msg.queueSize && parseInt(msg.queueSize) > 0) {
          queueSize = msg.queueSize;
        }

        verbose_log(msg.topic + " samplingInterval " + interval + " queueSize " + queueSize);
        verbose_log("Monitoring value: " + msg.topic + ' by interval of ' + interval.toString() + " ms");
```

mikakaraila, 2022-09-28:
Yes, you found it!

vdwpsmt, 2022-09-28:
So is my conclusion correct that in the case of a single subscription (action = subscribe and topic = nodeId), in the current situation,  one can only set the **publishinterval** (by using msg.interval or msg.payload) and not the **samplinginterval**?  Is that a missing feature?  It seems to be in conflict with what the documentation states.

mikakaraila, 2022-09-28:
SamplingInterval is used only with monitoredItem (client), server will decide publishInterval.

    async function subscribe_monitoredItem(subscription, msg) {
      verbose_log("Session subscriptionId: " + subscription.subscriptionId);

      // Simplified 
      if (msg.topic === "multiple") {
        verbose_log("Create monitored itemGroup for " + JSON.stringify(msg.payload));
        // let interval = 1000; // Default interval 
        let interval = opcuaBasics.calc_milliseconds_by_time_and_unit(node.time, node.timeUnit);
        if (msg && msg.interval) {
          interval = parseInt(msg.interval);
        }
        const monitorItems = [];
        for (var i=0; i<msg.payload.length; i++) {
          monitorItems.push({attributeId: AttributeIds.Value, nodeId: msg.payload[i].nodeId});
        }
        verbose_log("Using samplingInterval:" + interval);
        const monitoringParameters = {
          // clientHandle?: UInt32;
          samplingInterval: interval, // read from msg.interval
          // filter?: (ExtensionObject | null);
          queueSize: 1,
          discardOldest: true
        };

vdwpsmt, 2022-10-14:
ok, I understand that publishinterval is common and equal for all items in a subscription.  But is it possible to set an individual samplingInterval for each item?  If so, how?  I tried with individual messages, tried with "multiple", but can't seem to get it working in a consistent way.  this issue https://github.com/mikakaraila/node-red-contrib-opcua/issues/385 describes something similar and if I understand correctly from the last messages, a new feature has been added and it should be possible to specify the interval for each payload element (nodeId) individually (msg.payload[0].interval; msg.payload[1].intervall) .  Can someone confirm this is the case?  I could not find this in the documentation, or in the code...

mikakaraila, 2025-06-27:
I expect this can be closed. OPC UA does not provide individual parameters instead subscription contains those.

closed 2025-06-27 by mikakaraila