mikakaraila/node-red-contrib-opcua #483: OPCUa Browser Node Config. Name not Working

nobrefons, 2022-09-27:
Hello,

Node OPCUA Browser is not working with 'name' configuration.

Even adding the name 'OPC UA Browser' is always active:

[image]

[image]

Looking at the configuration on node and I saw that the configuration is this and I believe that configuration is missing:
[image]

I believe it would have to be in this pattern
[image]

mikakaraila, 2022-09-28:
Cosmetic, not effecting to functionality...

nobrefons, 2022-09-28:
Yes you are right.
For a next update if necessary.

closed 2022-09-28 by nobrefons