mikakaraila/node-red-contrib-opcua #484: Customization of debugger log

FabioLugatoDerga, 2022-09-29:
Hello there, I read multiple topics on these warnings, but I still can't remove those ones. 
[image]

I understand they are useful but there may be a way to turn them on and off without restarting node-red, maybe only redeploying? I really need to read other stuff on the logger.

piwi447, 2022-09-29:
I have same problem with warning 
`OPC SUBSCRIBE: StatusCode: Good#Overflow (0x00000080)`

coming each time my OP subscribe get update of one varabale and this is filling my debug windows but also the log file....
Thanks a lot

mikakaraila, 2022-09-30:
As long as status != StatusCodes.Good it will be logged.

I could change it to !goodish(status), but then you will not see it if it is like Good#Overflow. Do you need this?

```JavaScript
function isGoodish(statusCode: StatusCode): boolean {
  return statusCode.value < 0x40000000;
}
```

FabioLugatoDerga, 2022-09-30:
I don't want to turn off all logging but it would be great if the logging could be enabled/disabled with a checkbox.
Because yes the logging is essential but sometimes it clogs up the debug panel.
That may be a future feature if you consider it to be useful, for me it would be!

piwi447, 2022-09-30:
What is the exact meaning of good overflow? I could not find this status code in OPC UA status code list… is there something I can do to avoid this overflow message.
Basically my OPCUA communication is working well but generate a lot of error message and warning 

Pierre RUWET

mikakaraila, 2022-10-03:
It is used for the application level. Let´s say that AnalogDataItem has Range from 0 to 5 bar.
Overflow bit can be used to tell that measurement is 6 (over range). No OPC UA error, instead application level information.

FabioLugatoDerga, 2022-10-04:
> Put it to node-red wishlist, code is using node-red logging. It can be changed in the settings.js by logLevel.
> 
> ```js
>  // Configure the logging output
>     logging: {
>         // Only console logging is currently supported
>         console: {
>             // Level of logging to be recorded. Options are:
>             // fatal - only those errors which make the application unusable should be recorded
>             // error - record errors which are deemed fatal for a particular request + fatal errors
>             // warn - record problems which are non fatal + errors + fatal errors
>             // info - record information about the general running of the application + warn + error + fatal errors
>             // debug - record information which is more verbose than info + info + warn + error + fatal errors
>             // trace - record very detailed logging + debug + info + warn + error + fatal errors
>             // off - turn off all logging (doesn't affect metrics or audit)
>             level: "debug",
>             // Whether or not to include metric events in the log output
>             metrics: false,
>             // Whether or not to include audit events in the log output
>             audit: false
>         }
>     },
> ```

Ok but as you know this only affects console logging, not the node-red panel for logging. I already turned it all off.
What wishlist?

mikakaraila, 2022-10-04:
This seems to be already open on nodered forum... of course errors cannot be turned off.
If nodeId is incorrect it should be fixed. You have wrong namespace index ns=1 when it should be perhaps ns=3.
Or index number is wrong ns=4;i=1001 as it should be ns=4;i=1007 same with other String or Opaque nodeId formats.

piwi447, 2022-10-04:
Hello, 
I have this case with some structure I am reading.
This structure is a combination of 4 float values, 1 integer and one string.
This is seen by browser as extension object.
The subscribe is working and I am able to decode field names and the values but it generate error message complaining about opaque structure.
How can I configure the subscribe to avoid this kind of error message.
I saw an example with the build mode of OPC client but did not catch how to use it with a subscribe fonction
Thanks for any help

Pierre RUWET

mikakaraila, 2022-10-05:
Most probably it´s still inside node-opcua. 
There are still open issues there and root cause is somehow related to these dynamic extension objects.
You can check them from the node-opcua issue list.

piwi447, 2022-10-05:
Thanks for the feedback… I saw some discussion about dynamic extension object but I thought I was not in that case because my structure definition is static.So my extension object structure is also static.
I don’t understand the subscribe claiming about opaque structure when the structure can be decoded easily. 

Pierre RUWET

piwi447, 2022-10-05:
> Thanks a lot , then I need to investigate how to configure that range on the OPC UAserver side that is a codesys soft plc running on raspberry target.
>
> Pierre RUWET

I did not find any way to configure a range for my analog values on the server side.
If the OPC client claim for overflow, it probably compare the received value to limits.
These limits are thus probably defined on client side. 
My next question is then, where? And what do I need to add in the subscribe configuration to inform OPC client of the right range for each value?
Thanks for your support

Pierre RUWET

mikakaraila, 2022-10-05:
I expect overflow is generated on server side. 

For the extension object structure, can you add screenshot of the structure/variables? 

It could help me / Etienne to solve problem... it can be solved with the latest node-opcua as Etienne added some v1.03 nodeset emulation into the node-opcua.

piwi447, 2022-10-06:
Hello,

Regarding the Overflow, i tried to anlyse with OPC Expert

The Quality given is always good but wit code 192 which is different 
from àx00000080 generated by the opc client node.

piwi447, 2022-10-10:
Hello,
I worked teseterday again on more investigation about the overflow status.
i saw by logging the status code in the debug windoww that the overflow signal is not linked to a specifict item.
when the subscribe produce a value a message with a value in the payload and a topic related to one sunscribed item , the status code is always Good 0x00000000.
when the subscribe produce a status code overflow, there is no topic in the message

piwi447, 2022-10-11:
Thanks a lot , then I need to investigate how to configure that range on the OPC UAserver side that is a codesys soft plc running on raspberry target.

Pierre RUWET

mikakaraila, 2023-05-04:
Just a reminder that node-opcua uses own logging that I cannot effect.

You should give some examples which messages are not possible to "catch" as I have added those I think are critical.

FabioLugatoDerga, 2023-05-05:
These are some examples:
[image]
_"subscribe: StatusCode: Bad (0x80000000) The operation failed."_
_and also "Can't work with OPC-UA session"_ (when it won't reconnect on its own, not a big matter now since there are not many errors like this)

I understand that my specific use case produces several warnings that I can't avoid; these issues are on my end and not a fault of your code, but I was wondering if you could provide a way to catch/suppress these warnings so they don't appear in the debug sidebar. The OPCUA server on the machines sometimes returns an error even when everything is ok.