mikakaraila/node-red-contrib-opcua #485: Windows 10, What is the proper way to install openssl for node-red-contrib-opcua?

youngkzy, 2022-09-29:
@mikakaraila ,
I downloaded openssl-1.0.2o-x64_86-win64.exe for Windows 10 64 bit VM. I log in as machine root administrator and ran Node-RED with this file in the Node-RED directory as per the folder was opened by the system. I unzipped the file to this folder. This solved the issue ONLY for machine root admin. Node-RED runs from SYSTEM since it is started via NSSM. However, If I start Node-RED under another user, you now need to have it in the proper folder as per the user.
```
Error: cannot find file C:\Users<userlogin>\AppData\Local\Programs\openssl\openssl.exe
openssl seems to be missing and need to be installed
```

I'm assuming the install was not proper. How is this recommended for proper install of openssl?
Thanks

mikakaraila, 2022-10-03:
If there is no Internet connection (node-opcua-pki uses it to install openssl normally), then you have to install it manually.

FelixWalterAG, 2022-12-06:
Hi @mikakaraila 
is there a way to change the path where the module searches for openssl.exe?

mikakaraila, 2022-12-07:
Actual npm package that is used: https://github.com/node-opcua/node-opcua-pki

So you should add feature request there...

mikakaraila, 2022-12-08:
Here is actual workaround: https://github.com/node-opcua/node-opcua/issues/1165

mikakaraila, 2023-04-13:
See also: https://github.com/node-opcua/node-opcua/issues/1268

closed 2023-06-28 by mikakaraila