mikakaraila/node-red-contrib-opcua #490: Cannot find dataType Definition ! with nodeId & not expected processed dataType

kurago-dev, 2022-10-20:
## Subject of the issue

Sometimes, when starting Node-RED this error appears:

```log
10:27:47.068Z :populate_data_type_manager_104:54    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=3017
 ...                                                    at <NR-PATH>node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:41:23
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (<NR-PATH>node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
10:27:47.069Z :resolve_dynamic_extension_obje:148   err TypeError: Cannot read properties of null (reading 'constructor')
 ...                                                    at <NR-PATH>node_modules\node-opcua-client-dynamic-extension-object\dist\resolve_dynamic_extension_object.js:50:28
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (<NR-PATH>node_modules\node-opcua-client-dynamic-extension-object\dist\resolve_dynamic_extension_object.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
10:27:47.070Z :resolve_dynamic_extension_obje:149   opaqueStructure =  ns=2;i=5009
10:27:47.072Z :resolve_dynamic_extension_obje:150   opaqueStructure =  0x00000000: [error HEX]
 ...                                                
 ...                                                 .... ( 139)
10:27:47.073Z :resolve_dynamic_extension_obje:151   00000000: [error HEX]
 ...                                                
10:27:47.074Z :resolve_dynamic_extension_obje:152   ExtraDataTypeManager
 ...                                                ------------- namespace: http://opcfoundation.org/UA/
 ...                                                ------------- namespace: urn:ppc02-103:<namespace>:<namespace2>:OpcUaServer
 ...                                                
 ...                                                ------------- namespace: http://<namespace>.com/<namespace2>/
 ...                                                
```

Another error (that I think it's related) is:

```log
 08:25:21.024Z :resolve_dynamic_extension_obje:148   err Error: Internal Error: we are not expecting this dataType to be processed already
 ...                                                    at <NR-Path>\node_modules\node-opcua-client-dynamic-extension-object\dist\resolve_dynamic_extension_object.js:46:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (<NR-Path>\node_modules\node-opcua-client-dynamic-extension-object\dist\resolve_dynamic_extension_object.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
08:25:21.025Z :resolve_dynamic_extension_obje:149   opaqueStructure =  ns=2;i=5015
08:25:21.027Z :resolve_dynamic_extension_obje:150   opaqueStructure =  0x00000000: [Error HEX]
 ...                                                
 ...                                                 .... ( 172)
08:25:21.028Z :resolve_dynamic_extension_obje:151   00000000: [Error HEX]
 ...                                                
08:25:21.032Z :resolve_dynamic_extension_obje:152   ExtraDataTypeManager
 ...                                                ------------- namespace: http://opcfoundation.org/UA/
 ...                                                ------------- namespace: urn:ppc02-103:<namespace>:<namespace2>:OpcUaServer
 ...                                                
 ...                                                ------------- namespace: http://<namespace>.com/<namespace2>/
 ...                                                structureTypeName = <StructureName>
 ...                                                ERROR: cannot find constructor for encodingDefaultBinary
 ...                                                schema             name: <StructureName> (abstract= false )
 ...                                                        dataType NodeId: ns=2;i=3014
 ...                                                encoding Default Binary: ns=2;i=5003
 ...                                                encoding Default Xml   : ns=2;i=5004
 ...                                                name            <StructureName>
 ...                                                dataType        ns=2;i=3014
 ...                                                binaryEncoding  ns=2;i=5003
 ...                                                           <
 ...                                                           structVar1                     false type
 ...                                                           structVar2                     false type
 ...                                                           structVar3                     false type
                                                                >
 ...                                                 .... TRUNCATED ..... (NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE=25
```

## Your environment
- _Node-RED_ v3.0.2
- _node-red-contrib-opcua_ 0.2.289

### OPC-UA Server
This is a OPC-UA server exposing PLC variables. 

### Target variable(s)
The variables I am trying to subscribe to are Structures, so they're handled as `ExtensionObject`s if I understood correctly.

## Steps to reproduce

No idea, there doesn't seem to be a reason for this. The flow I am using is:

```mermaid
graph LR
    A[Inject] --> B(Function Block)
    B --> |Subscribe Multiple Nodes| C("OPC-UA Client (Subscribe)")
    C --> D[Debug]
```

Then sometimes one of the errors appears, sometimes for one of the subscribed variables, sometimes for another one and sometimes it doesn't appear at all. It seems arbitrary to me (multiple restarts without changing anything and different results).

The thing that bothers me is that whenever it works, it does so flawlessly, so I don't think the dataType has a bad implementation or can't be processed by `node-opcua`.

## Behaviour
When the `Cannot find dataType Definition ! with nodeId =` error appears, this is the output from the debug node in Node-RED:
[image]

Clicking again on the Inject node seems to fix it (sometimes), but it's not an ideal behaviour.

## Ideas
- This is related to #460, but it's not the same because in this case it's a Beckhoff PLC OPC-UA instead of a Siemens one, so Siemens' special DataTypes are not the issue.
  - However, on https://github.com/node-opcua/node-opcua/issues/1174, a user comments that he's using standard dataTypes from Siemens, so who knows.
- Maybe if an error like this appears the node could retry parsing the address space?

I really don't know what's going on but I am happy to provide any additional debug info.

mikakaraila, 2022-10-21:
I expect this one is related: https://github.com/node-opcua/node-opcua/issues/1107
So actual error is inside node-opcua not in the node-red-contrib-opcua.
@erossignon should take a look...

mikakaraila, 2023-01-24:
Please test with new v0.2.295 as it uses now node-opcua v2.89 that contains fixes to extension objects.

closed 2025-06-27 by mikakaraila