mikakaraila/node-red-contrib-opcua #491: Uncaught Exception while using browser node

aklauss, 2022-11-04:
Hello,

I've got this error while trying to use the browse node: 

4 Nov 08:04:52 - [error] [OpcUa-Browser:25b7a63fa762601b] Browse node error!
4 Nov 08:05:42 - [red] Uncaught Exception:
4 Nov 08:05:42 - [error] TypeError: NodeCrawler is not a constructor
    at browse_action_input (c:\users\xxxx\.node-red-1881\node_modules\node-red-contrib-opcua\opcua\102-opcuaclient.js:2021:25)
    at OpcUaClientNode.processInputMsg [as _inputCallback] (c:\users\xxxx\.node-red-1881\node_modules\node-red-contrib-opcua\opcua\102-opcuaclient.js:595:11)
    at c:\Users\xxxx\node_modules\@node-red\runtime\lib\nodes\Node.js:210:26
    at Object.trigger (c:\Users\xxxx\node_modules\@node-red\util\lib\hooks.js:166:13)
    at OpcUaClientNode.Node._emitInput (c:\Users\xxxx\node_modules\@node-red\runtime\lib\nodes\Node.js:202:11)
    at OpcUaClientNode.Node.emit (c:\Users\xxxx\node_modules\@node-red\runtime\lib\nodes\Node.js:186:25)
    at OpcUaClientNode.Node.receive (c:\Users\xxxx\node_modules\@node-red\runtime\lib\nodes\Node.js:485:10)
    at Immediate.<anonymous> (c:\Users\xxxxx\node_modules\@node-red\runtime\lib\flows\Flow.js:831:52)
    at processImmediate (node:internal/timers:466:21)

mikakaraila, 2022-11-04:
Have you updated manually node-opcua to latest one?

aklauss, 2022-11-04:
Nop. I will do that.

mikakaraila, 2022-11-04:
Actually it is not a good idea as I have followed node-opcua and crawler has been changed.
Most probably older version will work not the latest one.

aklauss, 2022-11-04:
Yes, I noticed that. I now doing a downgrade of node-opcua

mikakaraila, 2022-11-06:
I have node-opcua v2.77.0 in use and it is working.

erossignon, 2022-11-06:
The old crawler is being phased out and has been removed from "node-opcua" but  is still available for legacy compatibility, you just need to change the include from 
```
const { NodeCrawler  } = require("node-opcua");
```
to 
```
const { NodeCrawler  } = require("node-opcua-client-crawler");
```

mikakaraila, 2022-11-06:
Thanks, I will made change.

mikakaraila, 2022-11-06:
New version available v0.2.290

closed 2022-11-06 by mikakaraila