mikakaraila/node-red-contrib-opcua #492: OPC-UA Server is missing VariableValue

faulehenne, 2022-11-04:
Hey,

I'm trying to display Data i receive via MQTT to a NodeRed OPC-UA Server using this flow:

`[{"id":"cfb91e4987f2c163","type":"tab","label":"Flow 1","disabled":false,"info":"","env":[]},{"id":"9e58624.7faaba","type":"mqtt out","z":"cfb91e4987f2c163","name":"","topic":"esp32/output","qos":"","retain":"","broker":"10e78a89.5b4fd5","x":700,"y":680,"wires":[]},{"id":"abf7079a.653be8","type":"mqtt in","z":"cfb91e4987f2c163","name":"","topic":"esp32/temperature","qos":"2","datatype":"utf8","broker":"10e78a89.5b4fd5","nl":false,"rap":false,"rh":"0","inputs":0,"x":110,"y":340,"wires":[["cc79021b.9a751","21eae8f8.2971b8","2cfb0e72db40f97b"]]},{"id":"83cf37cf.c76988","type":"ui_switch","z":"cfb91e4987f2c163","name":"","label":"Output","group":"61285987.c20328","order":0,"width":0,"height":0,"passthru":true,"decouple":"false","topic":"","style":"","onvalue":"on","onvalueType":"str","onicon":"","oncolor":"","offvalue":"off","offvalueType":"str","officon":"","offcolor":"","x":90,"y":680,"wires":[["9e58624.7faaba"]]},{"id":"cc79021b.9a751","type":"debug","z":"cfb91e4987f2c163","name":"","active":true,"tosidebar":true,"console":true,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":360,"y":280,"wires":[]},{"id":"21eae8f8.2971b8","type":"ui_chart","z":"cfb91e4987f2c163","name":"","group":"61285987.c20328","order":0,"width":0,"height":0,"label":"Temperature","chartType":"line","legend":"false","xformat":"HH:mm:ss","interpolate":"linear","nodata":"","dot":false,"ymin":"","ymax":"","removeOlder":1,"removeOlderPoints":"","removeOlderUnit":"3600","cutout":0,"useOneColor":false,"useUTC":false,"colors":["#1f77b4","#aec7e8","#ff7f0e","#2ca02c","#98df8a","#d62728","#ff9896","#9467bd","#c5b0d5"],"outputs":1,"useDifferentColor":false,"className":"","x":350,"y":360,"wires":[[]]},{"id":"3680437c4c28f34d","type":"mqtt in","z":"cfb91e4987f2c163","name":"","topic":"esp32/pressure","qos":"2","datatype":"auto-detect","broker":"10e78a89.5b4fd5","nl":false,"rap":true,"rh":0,"inputs":0,"x":100,"y":200,"wires":[["22cf59c45ea91f6c"]]},{"id":"22cf59c45ea91f6c","type":"ui_chart","z":"cfb91e4987f2c163","name":"","group":"61285987.c20328","order":0,"width":0,"height":0,"label":"Pressure","chartType":"line","legend":"false","xformat":"HH:mm:ss","interpolate":"linear","nodata":"","dot":false,"ymin":"","ymax":"","removeOlder":1,"removeOlderPoints":"","removeOlderUnit":"3600","cutout":0,"useOneColor":false,"useUTC":false,"colors":["#1f77b4","#aec7e8","#ff7f0e","#2ca02c","#98df8a","#d62728","#ff9896","#9467bd","#c5b0d5"],"outputs":1,"useDifferentColor":false,"className":"","x":400,"y":200,"wires":[[]]},{"id":"2cfb0e72db40f97b","type":"OpcUa-Item","z":"cfb91e4987f2c163","item":"ns=1;s=PaulsTemp;datatype=String","datatype":"String","value":"\"Test\"","name":"Temperatur","x":350,"y":480,"wires":[["abd9b31982756ab9","ee410d56cb73180b"]]},{"id":"abd9b31982756ab9","type":"debug","z":"cfb91e4987f2c163","name":"debug 1","active":false,"tosidebar":true,"console":true,"tostatus":true,"complete":"payload","targetType":"msg","statusVal":"payload","statusType":"auto","x":650,"y":360,"wires":[]},{"id":"ee410d56cb73180b","type":"OpcUa-Server","z":"cfb91e4987f2c163","port":"4841","name":"NROPC","endpoint":"","users":"users.json","nodesetDir":"","folderName4PKI":"","autoAcceptUnknownCertificate":true,"registerToDiscovery":false,"constructDefaultAddressSpace":true,"allowAnonymous":true,"endpointNone":true,"endpointSign":true,"endpointSignEncrypt":true,"endpointBasic128Rsa15":true,"endpointBasic256":true,"endpointBasic256Sha256":true,"maxNodesPerBrowse":"1000","maxNodesPerHistoryReadData":0,"maxNodesPerHistoryReadEvents":0,"maxNodesPerHistoryUpdateData":0,"maxNodesPerRead":"1000","maxNodesPerWrite":"1000","maxNodesPerMethodCall":0,"maxNodesPerRegisterNodes":0,"maxNodesPerNodeManagement":0,"maxMonitoredItemsPerCall":0,"maxNodesPerHistoryUpdateEvents":0,"maxNodesPerTranslateBrowsePathsToNodeIds":0,"maxConnectionsPerEndpoint":"1000","maxMessageSize":4096,"maxBufferSize":4096,"x":640,"y":560,"wires":[["3e828e8b98926363"]]},{"id":"55f96a62301a0227","type":"inject","z":"cfb91e4987f2c163","name":"addVariable","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"1","crontab":"","once":true,"onceDelay":"5","topic":"ns=1;s=PaulsTemp;datatype=String","payload":"{ \"opcuaCommand\": \"addVariable\"}","payloadType":"json","x":130,"y":480,"wires":[["2cfb0e72db40f97b"]]},{"id":"3e828e8b98926363","type":"debug","z":"cfb91e4987f2c163","name":"debug 3","active":false,"tosidebar":true,"console":true,"tostatus":true,"complete":"payload","targetType":"msg","statusVal":"payload","statusType":"auto","x":870,"y":560,"wires":[]},{"id":"10e78a89.5b4fd5","type":"mqtt-broker","name":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"61285987.c20328","type":"ui_group","name":"Main","tab":"e7c46d5e.a1283","order":1,"disp":true,"width":"6","collapse":false},{"id":"e7c46d5e.a1283","type":"ui_tab","name":"Dashboard","icon":"dashboard"}]`

The debugger for my OPC-UA Server is giving me this error message: "warning: properties like messageType, namespace,
variableName or VariableValue is missing." and my Client (UAExpert) is showing the Variable with no Value at all. I can't figure out how to tell the OPC-UA Server where to look for the value it is missing. Thanks in advance.

mikakaraila, 2022-11-06:
Here is another example how server can use variables directly without OPC UA client node:

[{"id":"11a578056fc3d56a","type":"tab","label":"Flow 1","disabled":false,"info":"","env":[]},{"id":"8a4b4e742a3916ec","type":"OpcUa-Server","z":"11a578056fc3d56a","port":"53881","name":"","endpoint":"","users":"users.json","nodesetDir":"","folderName4PKI":"","autoAcceptUnknownCertificate":true,"registerToDiscovery":false,"constructDefaultAddressSpace":true,"allowAnonymous":true,"endpointNone":true,"endpointSign":true,"endpointSignEncrypt":true,"endpointBasic128Rsa15":true,"endpointBasic256":true,"endpointBasic256Sha256":true,"maxNodesPerBrowse":0,"maxNodesPerHistoryReadData":0,"maxNodesPerHistoryReadEvents":0,"maxNodesPerHistoryUpdateData":0,"maxNodesPerRead":0,"maxNodesPerWrite":0,"maxNodesPerMethodCall":0,"maxNodesPerRegisterNodes":0,"maxNodesPerNodeManagement":0,"maxMonitoredItemsPerCall":0,"maxNodesPerHistoryUpdateEvents":0,"maxNodesPerTranslateBrowsePathsToNodeIds":0,"x":640,"y":140,"wires":[["24ece21aaa3a8d8a"]]},{"id":"2f89c79ea33f7976","type":"inject","z":"11a578056fc3d56a","name":"add TestVariable","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"ns=1;s=TestVariable;datatype=Boolean","payload":"{\"opcuaCommand\": \"addVariable\"}","payloadType":"json","x":300,"y":40,"wires":[["8a4b4e742a3916ec"]]},{"id":"24ece21aaa3a8d8a","type":"debug","z":"11a578056fc3d56a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":940,"y":140,"wires":[]},{"id":"dcd3c29064d3f48c","type":"inject","z":"11a578056fc3d56a","name":"Set TestVariable false","props":[{"p":"payload.messageType","v":"Variable","vt":"str"},{"p":"payload.variableName","v":"TestVariable","vt":"str"},{"p":"payload.datatype","v":"Boolean","vt":"str"},{"p":"payload.namespace","v":"1","vt":"str"},{"p":"payload.variableValue","v":"false","vt":"bool"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":280,"y":100,"wires":[["8a4b4e742a3916ec"]]},{"id":"957057e20f53d587","type":"inject","z":"11a578056fc3d56a","name":"add TestVariable2 (Float)","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"ns=1;s=TestVariable2;datatype=Float","payload":"{\"opcuaCommand\": \"addVariable\"}","payloadType":"json","x":290,"y":460,"wires":[["8a4b4e742a3916ec"]]},{"id":"0b118afe42a49f53","type":"inject","z":"11a578056fc3d56a","name":"Set TestVariable2 12.3 + ts + quality (Number)","props":[{"p":"payload.messageType","v":"Variable","vt":"str"},{"p":"payload.variableName","v":"TestVariable2","vt":"str"},{"p":"payload.datatype","v":"Float","vt":"str"},{"p":"payload.namespace","v":"1","vt":"str"},{"p":"payload.variableValue","v":"12.3","vt":"str"},{"p":"payload.sourceTimestamp","v":"2022-05-25T20:36:00Z","vt":"str"},{"p":"payload.quality","v":"2161770496","vt":"num"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":250,"y":540,"wires":[["8a4b4e742a3916ec"]]},{"id":"8536d55d0424da30","type":"inject","z":"11a578056fc3d56a","name":"Set TestVariable \"true\"","props":[{"p":"payload.messageType","v":"Variable","vt":"str"},{"p":"payload.variableName","v":"TestVariable","vt":"str"},{"p":"payload.datatype","v":"Boolean","vt":"str"},{"p":"payload.namespace","v":"1","vt":"str"},{"p":"payload.variableValue","v":"true","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":280,"y":200,"wires":[["8a4b4e742a3916ec"]]},{"id":"a80b65bac7031056","type":"inject","z":"11a578056fc3d56a","name":"Set TestVariable 0 (false)","props":[{"p":"payload.messageType","v":"Variable","vt":"str"},{"p":"payload.variableName","v":"TestVariable","vt":"str"},{"p":"payload.datatype","v":"Boolean","vt":"str"},{"p":"payload.namespace","v":"1","vt":"str"},{"p":"payload.variableValue","v":"0","vt":"num"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":270,"y":300,"wires":[["8a4b4e742a3916ec"]]},{"id":"8e1d7776f2447bf4","type":"inject","z":"11a578056fc3d56a","name":"Set TestVariable 1 (true)","props":[{"p":"payload.messageType","v":"Variable","vt":"str"},{"p":"payload.variableName","v":"TestVariable","vt":"str"},{"p":"payload.datatype","v":"Boolean","vt":"str"},{"p":"payload.namespace","v":"1","vt":"str"},{"p":"payload.variableValue","v":"1","vt":"num"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":280,"y":340,"wires":[["8a4b4e742a3916ec"]]},{"id":"4dd1d84b6c01f3f3","type":"inject","z":"11a578056fc3d56a","name":"Set TestVariable true","props":[{"p":"payload.messageType","v":"Variable","vt":"str"},{"p":"payload.variableName","v":"TestVariable","vt":"str"},{"p":"payload.datatype","v":"Boolean","vt":"str"},{"p":"payload.namespace","v":"1","vt":"str"},{"p":"payload.variableValue","v":"true","vt":"bool"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":290,"y":140,"wires":[["8a4b4e742a3916ec"]]},{"id":"f427cfa2eb088e8a","type":"inject","z":"11a578056fc3d56a","name":"Set TestVariable \"false\"","props":[{"p":"payload.messageType","v":"Variable","vt":"str"},{"p":"payload.variableName","v":"TestVariable","vt":"str"},{"p":"payload.datatype","v":"Boolean","vt":"str"},{"p":"payload.namespace","v":"1","vt":"str"},{"p":"payload.variableValue","v":"false","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":280,"y":240,"wires":[["8a4b4e742a3916ec"]]},{"id":"eaeea5654fe61dce [... cut ...] payload.messageType","v":"Variable","vt":"str"},{"p":"payload.variableName","v":"TestVariable2","vt":"str"},{"p":"payload.datatype","v":"Float","vt":"str"},{"p":"payload.namespace","v":"1","vt":"str"},{"p":"payload.variableValue","v":"34.5","vt":"str"},{"p":"payload.sourceTimestamp","v":"2022-05-25T23:47:00Z","vt":"str"},{"p":"payload.quality","v":"UncertainInitialValue","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":250,"y":580,"wires":[["8a4b4e742a3916ec"]]}]

faulehenne, 2022-11-06:
Hey @mikakaraila,
thank you for the help but i can't import this code due to a syntax error. NodeRed is complaining about: "SyntaxError: JSON Parse error: Expected '}'.

mikakaraila, 2022-11-06:
Here as attachment, just rename .txt back to .json
[OPCUA-MQTT.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9945724/OPCUA-MQTT.txt)
[OPCUA-VARIABLE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9945725/OPCUA-VARIABLE.txt)

faulehenne, 2022-11-06:
Thank you very much. The first solution "OPCUA-MQTT"works for me. For any reason your inject (msg.payload = 23.5) didn't output anything on my side. But after changing the constant inject you created to my MQTT Input i can see the data in UAExpert. I'd like to create a flow without using the client to write to server, but the second piece of code you sent does not create any variables. The Server is working.

mikakaraila, 2022-11-06:
You have to push manually each inject node. Study flow so you can understand how variables can be used with server.

[image]

faulehenne, 2022-11-07:
Thanks, this helped me a lot. Any idea on how i can "inject" the MQTT input as  "msg.payload.variableValue" for the server?

mikakaraila, 2022-11-07:
Use function node to add other needed fields into the msg object or payload.

closed 2022-11-07 by mikakaraila