mikakaraila/node-red-contrib-opcua #498: Sending Description Attribute when adding variable

AndreSchm, 2022-11-22:
Is it possible to set the Description attribute when adding a variable to the address space?
I tried something like
`msg.payload = { "opcuaCommand": "addVariable" }; msg.topic = 'ns=1;s=MyTestInt;datatype=UInt32;description=test'`
but without any effect.

mikakaraila, 2022-11-28:
Yes, it will need just a small code to make it possible, next version.

mikakaraila, 2022-12-04:
I added feature to both addFolder and addVariable.

addFolder, description: MyTestFolder
[image]

addVariable, decription: MyVariableX
[image]

AndreSchm, 2022-12-09:
Thank you very much 👍

closed 2022-12-09 by AndreSchm