mikakaraila/node-red-contrib-opcua #500: Connection failed,and it make node-red crash

Jireyo, 2022-12-07:
Condition:
A simple opcua-client node,it connect to a PLC(s7-1200) with "None" security policy, "None" security mode , use anonymous.

Phenomenon:
As I start my node-red,usually it will connect to the PLC automatually and display a "session active" under the node,but sometime it 
 failed,and the node-red would crash.
It's deadly,now it always happens and I can't even open visual interface to change flow and avoid it.（node-red start and crash immediately）

Here is console displayed:
`
2022-12-08 01:43:09: 17:43:09.609Z :client_secure_channel_layer   :206   requestData not found for requestId =  5 try with  6
2022-12-08 01:43:09: 8 Dec 01:43:09 - [red] Uncaught Exception:
2022-12-08 01:43:09: 8 Dec 01:43:09 - [error] TypeError: Cannot read property 'callback' of undefined
2022-12-08 01:43:09:     at MessageBuilder.<anonymous> (C:\Users\Administrator\.node-red\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:213:42)
2022-12-08 01:43:09:     at MessageBuilder.emit (events.js:400:28)
2022-12-08 01:43:09:     at MessageBuilder._report_error (C:\Users\Administrator\.node-red\node_modules\node-opcua-transport\dist\source\message_builder_base.js:128:14)
2022-12-08 01:43:09:     at MessageBuilder._feed_messageChunk (C:\Users\Administrator\.node-red\node_modules\node-opcua-transport\dist\source\message_builder_base.js:193:22)
2022-12-08 01:43:09:     at PacketAssembler.<anonymous> (C:\Users\Administrator\.node-red\node_modules\node-opcua-transport\dist\source\message_builder_base.js:64:66)
2022-12-08 01:43:09:     at PacketAssembler.emit (events.js:400:28)
2022-12-08 01:43:09:     at PacketAssembler.feed (C:\Users\Administrator\.node-red\node_modules\node-opcua-packet-assembler\dist\packet_assembler.js:75:18)
2022-12-08 01:43:09:     at MessageBuilder.feed (C:\Users\Administrator\.node-red\node_modules\node-opcua-transport\dist\source\message_builder_base.js:94:35)
2022-12-08 01:43:09:     at ClientSecureChannelLayer._on_receive_message_chunk (C:\Users\Administrator\.node-red\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:913:29)
2022-12-08 01:43:09:     at ClientTCP_transport.<anonymous> (C:\Users\Administrator\.node-red\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:758:18)
`
I know maybe there is some trouble with the OPCUA server,but it's necessary to avoid crashes of the process.

thalesmaoa, 2023-02-23:
Not sure if its related, but if I use OPC node with external topic and doesn't define datatype, it also crashes.

mikakaraila, 2023-07-04:
No activity => closed. Or provide more information if you want to reopen.

closed 2023-07-04 by mikakaraila