mikakaraila/node-red-contrib-opcua #501: Reconnect doesnt work

Kaktor, 2022-12-09:
We subscribed different OPC-UA Tags with node-red-contrib-opcua 0.2.292.
Sometime the node losing the connection to the OPC-UA Server because a network issue or the machine went offline - after that
node-red flow stuck and we have to do a manually re-deploy. 

In logs are following message

`
2022-12-08 03:20:11	
02:20:11.514Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:19:51	
02:19:51.512Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:19:31	
02:19:31.513Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:19:11	
02:19:11.511Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:18:51	
02:18:51.509Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:18:31	
02:18:31.508Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:18:11	
02:18:11.506Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:17:51	
02:17:51.503Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:17:31	
02:17:31.502Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:17:11	
02:17:11.502Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:17:10	
02:17:10.630Z :client_secure_channel_layer   :884   Invalid socket => Communication has been lost, cannot renew token
	
2022-12-08 03:16:51	
02:16:51.500Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:16:31	
02:16:31.499Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:16:11	
02:16:11.497Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:15:51	
02:15:51.494Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:15:31	
02:15:31.494Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:15:11	
02:15:11.491Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:14:51	
02:14:51.488Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest
	
2022-12-08 03:14:31	
02:14:31.487Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms , request = ReadRequest)`

mikakaraila, 2022-12-09:
I expect this causes it cannot reconnect:
02:17:10.630Z :client_secure_channel_layer   :884   Invalid socket => Communication has been lost, cannot renew token

Normally node-opcua takes care of reconnect. This is some special case, if you need some professional support contact: @erossignon

mikakaraila, 2022-12-21:
Cannot fix network error that causes state that node-opcua built-in reconnect cannot solve.

mikakaraila, 2023-01-24:
No updates => closed

closed 2023-01-24 by mikakaraila