mikakaraila/node-red-contrib-opcua #503: DataType Uint16 BadTypeMismatch (0x80740000)

simo-fol, 2022-12-19:
Hello,
I am trying to write into a Panasonic PLC some values. I can write in many nodes except for the UInt16.

The example node is the following and the DataType is UInt16, but the value looks like a UInt16Array:

x-special/nautilus-clipboard
copy
file:///home/soltech/Scrivania/uaexpert.png

I tried different ways to write but I always get the same error:

x-special/nautilus-clipboard
copy
file:///home/soltech/Scrivania/nodered.png

I'm not very practical so sorry in advance!

[{"id":"b984566bfc333873","type":"tab","label":"Flow 2","disabled":false,"info":"","env":[]},{"id":"2cf2817572273bf9","type":"OpcUa-Item","z":"b984566bfc333873","item":"ns=2;s=Panasonic FP/KW.Tags.NOTE_RICETTA_9","datatype":"UInt16","value":"5","name":"UInt16","x":630,"y":180,"wires":[["9c99ef3ab773df49"]]},{"id":"21eadc9a8f220ffd","type":"OpcUa-Client","z":"b984566bfc333873","endpoint":"b1cf60f415500f24","action":"write","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","folderName4PKI":"","name":"Write","x":830,"y":260,"wires":[["f69ba1cd52f7b02c"]]},{"id":"14c65132bb827d72","type":"inject","z":"b984566bfc333873","name":"Test","props":[{"p":"payload","v":"","vt":"str"},{"p":"topic","v":"","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":"","topic":"","payload":"","payloadType":"str","x":430,"y":180,"wires":[["2cf2817572273bf9"]]},{"id":"07d55e95a92aaad1","type":"OpcUa-Item","z":"b984566bfc333873","item":"ns=2;s=Panasonic FP/KW.Tags.NOTE_RICETTA_9","datatype":"UInt16 Array","value":"5,4,3,2,16","name":"UInt16Array","x":650,"y":260,"wires":[["21eadc9a8f220ffd"]]},{"id":"1372e15688a9a549","type":"debug","z":"b984566bfc333873","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","x":1310,"y":220,"wires":[]},{"id":"f69ba1cd52f7b02c","type":"function","z":"b984566bfc333873","name":"Get StatusCode value","func":"var status = msg.payload.toJSONFull();\nvar status2 = JSON.parse(JSON.stringify(status));\n\nnode.warn(\"Value:\" + status2.value);\nnode.warn(\"Name: \" + status2.name);\nnode.warn(\"Desc: \" + status2.description);\n\n// Return just statuscode value\nmsg.payload = status2.value;\n\nreturn msg;","outputs":1,"noerr":0,"x":1080,"y":220,"wires":[["1372e15688a9a549"]]},{"id":"9c99ef3ab773df49","type":"OpcUa-Client","z":"b984566bfc333873","endpoint":"b1cf60f415500f24","action":"write","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","folderName4PKI":"","name":"Write","x":830,"y":180,"wires":[["f69ba1cd52f7b02c"]]},{"id":"9a2e4d14b69bd58f","type":"inject","z":"b984566bfc333873","name":"Test","props":[{"p":"payload","v":"","vt":"str"},{"p":"topic","v":"","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":"","topic":"","payload":"","payloadType":"str","x":430,"y":260,"wires":[["07d55e95a92aaad1"]]},{"id":"b1cf60f415500f24","type":"OpcUa-Endpoint","endpoint":"opc.tcp://192.168.1.130:4840","secpol":"None","secmode":"None","none":true,"login":false,"usercert":false,"usercertificate":"","userprivatekey":""}]

mikakaraila, 2022-12-19:
I cannot test with your server, but here is example flow with Prosys UA Simulation Server:
[image]

[EXAMPLE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10259450/EXAMPLE.txt)

simo-fol, 2022-12-19:
No, it does not work, always the same error.
[image]

mikakaraila, 2022-12-19:
Can you write one value or will it require the whole array write with UaExpert?

Prosys Simulation Server array is for 5 elements and your is for 13 elements.
[image]

simo-fol, 2022-12-19:
Finally i made it! Thank you.
I have to pass each element of the array. I used your example "OPCUA-TEST-NODES.json" in "" Test Types (write)" tab and i think you should fix your "Uint16 Array" OPC-UA Item because the values are "5,4,3,2,1" instead of "55,44,33,22,11" (I'm not expert so maybe i'm wrong!).

Now the problem is that i have a string like "Hello there" to write into the UInt16Array node. How can I convert a string to UInt16Array?

mikakaraila, 2022-12-19:
Here, updated example:
[image]

[EXAMPLE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10261852/EXAMPLE.txt)

simo-fol, 2022-12-20:
Thank you! 
But how can i create the array I need? I mean, with this method I get one integer for each character, in this case 11 integers, how can I convert it into 11,22,33?

 in this case I have to write "Hello world" into a UInt16Array of 3 elements, how can I do it?

simo-fol, 2022-12-20:
Thank you! 
But how can i create the array I need? I mean, with this method I get one integer for each character, for "Hello world" I get 11 integers, how can I convert it into a UInt16Array of certain number of elements?

mikakaraila, 2022-12-20:
It depends on your server Array length, you can always put into the while loop some max length after it will break string conversion.

If you use node-red embedded OPC UA Server then you can create Array with needed length before you write it, but I expect this is not the case.

simo-fol, 2022-12-20:
What can I do if my string length is longer than my array number of elements? Like "Hello world" for an array of 3 elements?

mikakaraila, 2022-12-20:
Then it will not fit, server decides array size and it will most probably give some error if you try to overwrite existing buffer length.

simo-fol, 2022-12-20:
I finally made it!
I had to convert your function result (8-bit) into a 16-bit, for everyone's wondering how to convert it:

> var a = 49
> var b = 50
> var res = ((b << 8) | a))        
>
>  --> 12849

Hope that helps!

Thank you so much for your time!

closed 2022-12-20 by simo-fol