mikakaraila/node-red-contrib-opcua #505: OPC UA Multiple Interval settings while using Subscription

Mani-i4point0, 2022-12-22:
hi,
In my project, I was trying to set the OPC UA Subscription with two intervals using specific variable value [ true ] from the read information received from the opc ua Client. and as soon as the value of variable changes to false need to have a subscription with the longer interval.
I am using the nodes from 'node-red-contrib-opcua'
Basically , would like to get a quicker update of the variables based on the one specific status, example in a normal scenario the subscription interval shall be 10seconds and based on the one of the variable state[true] the subscription interval needs to be changed to 1sec and then back to 10sec for the variable status has become false.

Tried to using msg.payload , no luck yet.

Can you please advise me on the nodes i need use to achieve two different intervals to the OPC UA Subscription node ?
On my flow, on the subscription method to read, I tried using template and CSV node to configure all the data in one place, but was ended up with an error, I think, Stringarray data type is creating a problem
Can you please suggest me in using the right method to use template and csv node for the Subscription method for a Stringarray  datatype  ?

[flows (1).json](https://discourse.nodered.org/uploads/short-url/lkLIDg8Zckl4gb7iRDfDliVgsXG.json) (101.2 KB)

You Valuable inputs highly appreciated.

Best Regards,
_Mani_

[https://discourse.nodered.org/t/opc-ua-multiple-interval-settings-while-using-subscription/71739](url)

mikakaraila, 2022-12-23:
Each subscription has own parameters => you have to use two subscriptions (2 client nodes with different parameters).

Mani-i4point0, 2022-12-26:
Thanks for response, the intension of having multiple subscription to reduce my CPU and RAM usage.
Having one more client node will increase the RAM ,CPU load and max number of sessions, etc...

It would be nice feature to have a configurable interval based on event, so that my CPU Load and RAM utilization goes high for a shorted period instead of all the time. 

Example : I would like to Monitor the Tank Level at a 200ms whenever the motor is ON, so that I can switch off the motor precisely before it overflows. Remaining all the time the subscription rate can be at 2sec.

Or Please do suggest any solution which can't eat-up my CPU Load and RAM

Best Regards,
Mani

mikakaraila, 2022-12-27:
Which forum you want to use for discussion? 

I don´t want to use both, it is useless to repeat same answer on both node-red forum and this one.

mikakaraila, 2022-12-27:
Closed this one as it is stupid to fill up both forum with the same issue.

Mani-i4point0, 2022-12-27:
> Which forum you want to use for discussion?
> 
> I don´t want to use both, it is useless to repeat same answer on both node-red forum and this one.

Sure @mikakaraila  : Will use Node-red forum. Thanks for responding to my queries.

closed 2022-12-27 by Mani-i4point0