mikakaraila/node-red-contrib-opcua #507: How to Read multiple variables in OpcUa

ahmedt04, 2023-01-02:
Hello, I have to read some variables through OpcUa but there are so many variables how can I read them using few nodes?
thank you.
`
[
    {
        "id": "fe571d3e2703ca9f",
        "type": "tab",
        "label": "Flow 44",
        "disabled": false,
        "info": ""
    },
    {
        "id": "a879dc9f7fe3d84d",
        "type": "OpcUa-Client",
        "z": "fe571d3e2703ca9f",
        "endpoint": "65f390f2746b4b99",
        "action": "readmultiple",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "name": "passo",
        "x": 290,
        "y": 200,
        "wires": [
            [
                "903985c37672f453",
                "30e54702bb70d0f5"
            ]
        ]
    },
    {
        "id": "1b067e0b73e7cc2e",
        "type": "inject",
        "z": "fe571d3e2703ca9f",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "0.25",
        "crontab": "",
        "once": true,
        "onceDelay": "1",
        "topic": "ns=3;s=\"DB HMI TRN GT1\".\"Hmi_In\".\"ParOperatore\";",
        "payloadType": "date",
        "x": 130,
        "y": 200,
        "wires": [
            [
                "a879dc9f7fe3d84d"
            ]
        ]
    },
    {
        "id": "903985c37672f453",
        "type": "OpcUa-Item",
        "z": "fe571d3e2703ca9f",
        "item": "ns=3;s=\"DB HMI TRN GT1\".\"Hmi_In\".\"ParOperatore\"",
        "datatype": "String",
        "value": "msg.payload",
        "name": "",
        "x": 520,
        "y": 180,
        "wires": [
            [
                "023e5265b1ee3939",
                "af33e5f21da99390"
            ]
        ]
    },
    {
        "id": "30e54702bb70d0f5",
        "type": "debug",
        "z": "fe571d3e2703ca9f",
        "name": "",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "payload",
        "statusType": "auto",
        "x": 450,
        "y": 380,
        "wires": []
    },
    {
        "id": "023e5265b1ee3939",
        "type": "OpcUa-Client",
        "z": "fe571d3e2703ca9f",
        "endpoint": "65f390f2746b4b99",
        "action": "write",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "name": "",
        "x": 900,
        "y": 140,
        "wires": [
            []
        ]
    },
    {
        "id": "af33e5f21da99390",
        "type": "debug",
        "z": "fe571d3e2703ca9f",
        "name": "",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "payload",
        "statusType": "auto",
        "x": 770,
        "y": 320,
        "wires": []
    },
    {
        "id": "65f390f2746b4b99",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://10.137.1.20:4840",
        "secpol": "None",
        "secmode": "None",
        "login": false
    }
]
`
[image]

mikakaraila, 2023-01-02:
Look examples and OPCUA-TEST-NODES.json

This tab and flow:
[image]

ahmedt04, 2023-01-02:
> Guarda esempi e OPCUA-TEST-NODES.json
> 
> Questa scheda e flusso: [image]

already looked at it and tried to use it,
the problem is that I have to read more than 300 variables
and all the variables are not in the same datablocks

mikakaraila, 2023-01-03:
I am normally using file and read nodeIDs from it.

This way:
[image]

Example flow:
[NODEIDS_FROM_FILE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10334541/NODEIDS_FROM_FILE.txt)

File:
[prosys.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10334546/prosys.txt)