mikakaraila/node-red-contrib-opcua #508: Error notification displays password in plain text

s8iva, 2023-01-06:
Hi

Its been noticed for certain error messages the username and password are displayed in plain text. E.g. "Read UA Server: Case B: UserName & password does not match to server (needed by Sign): engineer Welcome2Windows!". This is not secure, could you please introduce some obfuscation within the message to deal with this.

Thanks
Nikhil Mahesh

mikakaraila, 2023-01-09:
Yes, I made it visible on purpose. As normally console is not visible for end users.

Actually to make this secure passwords in the users.json file should be encrypted.

mikakaraila, 2023-02-12:
Removed password from the error message.

closed 2023-02-12 by mikakaraila

AndreSchm, 2023-02-22:
> Yes, I made it visible on purpose. As normally console is not visible for end users.
> 
> Actually to make this secure passwords in the users.json file should be encrypted.

Do you have a suggestion on how to encrypt users.json? 
I was thinking about using [node-red-contrib-credentials](https://github.com/Steveorevo/node-red-contrib-credentials) in combination with setUsers Command but I still get _BadUserAccessDenied_ after injection.
[image]
with msg.payload as
`{"opcuaCommand":"setUsers","users":[{"username":"test"},{"password":"test"},{"role":"AuthenticatedUser"}]}`
The Users file property is empty. Am I missing something?

mikakaraila, 2023-02-22:
Sorry but as this is just example how to make things working together, I don´t have time to make it more secure.
There are different ways to make password encryption. Another way is to used certificate that is user specific.
But to use it you will need some assistance from sterfive support.

AndreSchm, 2023-02-22:
I completely understand, but my setUsers call is correct is it?

mikakaraila, 2023-02-22:
You can test it by modifying users.json I expect code reads it by default.... have to check from code.