mikakaraila/node-red-contrib-opcua #509: OPCUA-Client "WRITE" : Error: The processing will complete asynchronously.

Daryldom123, 2023-01-06:
Hello,

Hope I haven't missed something in the issues list yet, but I'm running into an issue that I'm struggling to diagnose. 

I'm working on an application that reads data from an OPC server that connects multiple PLCs, node-red is essentially reading that data to then route it to where it's more useful, for testing that's just a node-red-dashboard. 

READ / SUBSCRIBE have all gone well once I seperately installed OpenSSL, but I've been struggling with "WRITE".

If I include the usual tag structure, nsu=[Namespace];ns=[X];s=[TagID], then nothing at all happens when I "WRITE", but "READ" works fine. 

If I remove the nsu=[NameSpace], then "READ" continues to work, but "WRITE" now generates an error at the OPCUA Client of "ERROR: The processing will complete asynchronously".

Which... I can't for the life of me figure out what exactly to do with that 
information.
Prior to any actions, but connection established:
[image]

After injecting node for "READ"
[image]

After injecting node for "WRITE"
[image]

Any help is appreciated.

mikakaraila, 2023-01-09:
Response is OPC UA error code. What server you are using?

Daryldom123, 2023-01-09:
Unfortunately, one that I have very little control over. It's the OPCUA server as configured by the PLC environment (Being a PILZ PSS4000 system). 

Is it something you've encountered before? I can write to the tags using OPCLabs QuickOPC Library, but not yet via node-red.

Daryldom123, 2023-01-09:
I was able to further diagnose the issue and determine that the cause for the failed write was in the PLC Variable structure. While the Client still returns an "Error: The process will complete asynchronously" it now correctly writes. 

Great work on this library, it's been impressively powerful for how simple it is

closed 2023-01-09 by Daryldom123