mikakaraila/node-red-contrib-opcua #510: Can't connection to Siemens S7-1500 OPC-UA Server with Certificate

bob3150, 2023-01-08:
I have a connection to my Siemens S7-1500 OPC-UA when SecurityPolicy and SecurityMode are set to None. But I have to secure my connection and want to use SecurityPolicy Basic256Sha256 and SecurityMode SignAndEncrypt. When I use this options with Procsys OPC UA Browser, I can establish a connection to my Siemens Server. But with node red not.
I use Procsys on a Mac which is connected over an SSH-Tunnel on the Ubuntu-machine where also Node-Red is running on it.
I'm not sure why my Siemens OPC-UA-Server is not accepting the certificate from Node-Red, but the one from Procsys.
Are Certificates bound to a Network-Name or IP-address? Because I use several IP-addresses on the Ubuntu-machine and the communication to my Siemens-Server has no hostnames, no DNS, no gateway, it's only IP-based. Can this be the problem?

mikakaraila, 2023-01-09:
Have you looked console messages? It can be that Siemens certificate contains only hostname.

To validate it you have to add IP & hostname into the your hosts file (if DNS is not used).

bob3150, 2023-01-09:
In the log the output is:

`9 Jan 21:18:57 - [warn] [OpcUa-Client:S7-1500 2] S7-1500 2: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:[3] securityPolicy:["http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256"]
`
`
9 Jan 21:18:57 - [warn] [OpcUa-Client:S7-1500 2] S7-1500 2: Case B: UserName & password does not match to server (needed by Sign): username password
`

in case of the host file, where I have to add the ip on my node-red host or on the Siemens. But on the Siemens-S7-1500 SPS I didn't know how to do this.

mikakaraila, 2023-01-10:
Are you using corrent security policy? And S7 OPC UA Server must trust/import client certificate.

@AndreasHeine you have perhaps some ready instructions for this?

AndreasHeine, 2023-01-10:
the only thing i know about it is that siemens requires a x509v3 certificate with the subject altname extension and that they check the application uri (but that might changed with newer fw...)
what is the actual opcua statuscode if you attempt to connect?

mikakaraila, 2023-01-11:
Or the Siemens S7 will need hostname of the client to validate certificate...

Have you looked Siemens documentation how to connect with certificate?

AndreasHeine, 2023-01-11:
it depends on the configuration of the plc... TIA-Portal has a way to manage the certificates (PKI)!

if the plc has a strict configuration you need to update the plc's trusted list...

sorry for the german settings... its the trusted client part in the opc ua configuration:
[image]

> I looked closed issues and this one could help:
> https://github.com/mikakaraila/node-red-contrib-opcua/issues/462

the solution for that is different -> the node-opcua-server needs to has the ip in serveroptions as "hostname" and domain-names as "alternatehostnames" due to the not existing dns resolve capabilities!

AndreasHeine, 2023-01-11:
the most important thing is what the actual statuscode is returned while connecting?

mikakaraila, 2023-01-11:
It should be visible on node itself:
 set_node_errorstatus_to("invalid endpoint", err);

Eldarion85, 2023-01-25:
I use this client with about 15 Siemens S7-1500. All are set to the highest security (Basic256Sha256). Guest access is disabled and user authentication is active. Also, no client certificates are automatically accepted, so the servers and client check each other's certificates.

With almost all S7-1500 the client certificates are not in the certificate memory of the CPU, only the CA certificates. Also, I use intermediate-CAs and the client sends the whole chain with its certificate.

There are no interruptions, the connections are stable and reestablish themselves automatically when the client or server is restarted.

But @bob3150 didn't answer here anymore, the problem has probably been solved.

closed 2023-01-25 by mikakaraila