mikakaraila/node-red-contrib-opcua #512: How to send variables to display in hmi siemens with modbus?

ahmedt04, 2023-01-19:
Hello,
i'm using node-red to get some variables from a plc via OpcUa i'm using an iot2050,
I managed to read the variables I need now I should send these variables to an hmi panel to be able to read them on the panel,
I have to do it in modbus, but I've never used modbus, does anyone know how to do it?
thank you.

mikakaraila, 2023-01-19:
For MODBUS look node-red-contrib-modbus

ahmedt04, 2023-01-19:
I already downloaded node-red-contrib-modbus
now after I read the variables taken from the plc in opcua I must be able to send them to a hni panel to be able to read them on the panel
and i have to do it using node-red-contrib-modbus but i wouldn't know how to do it.