mikakaraila/node-red-contrib-opcua #514: invalid endpoint error

acomtools, 2023-01-23:
Dear Support,

What went wrong? There is an error invalid endpoint error.  
here is the flow:

[
    {
        "id": "588932b17745db40",
        "type": "OpcUa-Client",
        "z": "76212d59e4af0514",
        "endpoint": "77c3523f9c89d6f2",
        "action": "read",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "l",
        "localfile": "C:\\PGR\\NodeRed\\PKI\\trusted\\certs\\NodeOPCUA-Client@aOEE[44233b688e247b3924184be5602aa622ac6b7f2f].pem",
        "localkeyfile": "C:\\PGR\\NodeRed\\PKI\\own\\private\\private_key.pem",
        "securitymode": "None",
        "securitypolicy": "None",
        "folderName4PKI": "C:\\PGR\\NodeRed",
        "name": "CPS-OPC-UA",
        "x": 280,
        "y": 100,
        "wires": [
            [
                "c8a06bb27ac22636"
            ]
        ]
    },
    {
        "id": "6c2254fa5a69f783",
        "type": "inject",
        "z": "76212d59e4af0514",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "ns=2;s=CPS-MG341-ADSC.Counter.CNT00;datatype=UInt32",
        "payload": "",
        "payloadType": "date",
        "x": 100,
        "y": 100,
        "wires": [
            [
                "588932b17745db40"
            ]
        ]
    },
    {
        "id": "c8a06bb27ac22636",
        "type": "debug",
        "z": "76212d59e4af0514",
        "name": "debug 7",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "x": 460,
        "y": 100,
        "wires": []
    },
    {
        "id": "77c3523f9c89d6f2",
        "type": "OpcUa-Endpoint",
        "z": "76212d59e4af0514",
        "endpoint": "opc.tcp://192.168.0.101:4840",
        "secpol": "None",
        "secmode": "None",
        "none": false,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]

[image]

mikakaraila, 2023-01-23:
Did you check connection parameters, try with Sign and username & password.

Which version you are using? There was initialization problem in some older version.

mikakaraila, 2023-01-23:
Have you checked Anonymous? If you are not providing username & password it must be checked.

[image]

acomtools, 2023-01-23:
I check the Anonymous.

 

Fyi: I try the same setting di dataFEED OPC-UA client, it work.

 

Best regards,

Karyadi

 

 <https://user-images.githubusercontent.com/5184524/213995230-d6ae6e89-9539-49e5-89f0-a1f54d804cfd.png>

acomtools, 2023-01-25:
02:00:14.512Z :message_chunker               :174   [NODE-OPCUA-E11] message size 65537 exceeds the negotiated message size 8192 nb chunks 1
02:00:14.516Z :client_secure_channel_layer   :848   { /*ServiceFault*/
 ...                                                 responseHeader                /*  ResponseHeader                            */: {
 ...                                                   timestamp                   /*  DateTime                                  */: 2023-01-25T02:00:14.513Z
 ...                                                   requestHandle               /*  UInt32                                    */: 0               0x0
 ...                                                   serviceResult               /*  StatusCode                                */: BadInternalError (0x80020000)
 ...                                                   serviceDiagnostics          /*  DiagnosticInfo                            */: { /*DiagnosticInfo*/
 ...                                                 namespaceUri                  /*  Int32                                     */: null
 ...                                                 symbolicId                    /*  Int32                                     */: -1
 ...                                                 locale                        /*  Int32                                     */: -1
 ...                                                 localizedText                 /*  Int32                                     */: -1
 ...                                                 additionalInfo                /*  String                                    */: null
 ...                                                 innerStatusCode               /*  StatusCode                                */: Good (0x00000000)
 ...                                                 innerDiagnosticInfo           /*  DiagnosticInfo                            */: null
 ...                                                };
 ...                                                   stringTable                 /*  String                                []  */: [ /* length =1*/
 ...                                                     message size exceeds the negotiated message size
 ...                                                   ]
 ...                                                   additionalHeader            /*  ExtensionObject                           */: null
 ...                                                 }
 ...                                                };
02:00:14.523Z :client_secure_channel_layer   :849   xxxxx  <<<<<< _on_message_received  ERROR requestId= 1 Object response.responseHeader.requestHandle= 0                   ServiceFault
25 Jan 09:00:14 - [warn] [OpcUa-Client:CPS-OPC-UA] CPS-OPC-UA: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:[1] securityPolicy:["http://opcfoundation.org/UA/SecurityPolicy#None"]
25 Jan 09:00:14 - [warn] [OpcUa-Client:CPS-OPC-UA] CPS-OPC-UA: Case B: UserName & password does not match to server (needed by Sign): undefined undefined

mikakaraila, 2023-01-25:
Have you tried to fine tune these parameters:
[image]

mikakaraila, 2023-04-23:
Closed as no more feedback

closed 2023-04-23 by mikakaraila