mikakaraila/node-red-contrib-opcua #515: No output in case of connection error

Abhijiht, 2023-01-23:
Hi,
I am using the OpcUa-Client node in read mode. An issue I'm facing is that it does not give any output when there is a connection issue. How can I get an output, even if it is an error message?

mikakaraila, 2023-01-23:
Perhaps you should attach your flow here, cannot give any meaningful comment without it.

Abhijiht, 2023-01-23:
Hi,
Here is the flow:
[
    {
        "id": "e5fe74e64e9875d3",
        "type": "OpcUa-Client",
        "z": "2808fa8a52f782a2",
        "endpoint": "b45abd73.56c1e7",
        "action": "read",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "name": "",
        "x": 460,
        "y": 160,
        "wires": [
            [
                "fc2c0b05bb8f72ae"
            ]
        ]
    },
    {
        "id": "f31a5e238c4114ce",
        "type": "inject",
        "z": "2808fa8a52f782a2",
        "name": "opcuaTrigger",
        "props": [
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "10",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "\"datatype=int32;ns=3;i=1001\"",
        "x": 250,
        "y": 160,
        "wires": [
            [
                "e5fe74e64e9875d3"
            ]
        ]
    },
    {
        "id": "fc2c0b05bb8f72ae",
        "type": "function",
        "z": "2808fa8a52f782a2",
        "name": "processData",
        "func": "\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 670,
        "y": 160,
        "wires": [
            [
                "e589590c7b8774a3"
            ]
        ]
    },
    {
        "id": "e589590c7b8774a3",
        "type": "mssqlServer",
        "z": "2808fa8a52f782a2",
        "name": "",
        "connStr": "",
        "x": 870,
        "y": 160,
        "wires": []
    },
    {
        "id": "b45abd73.56c1e7",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://localhost:53530/OPCUA/SimulationServer",
        "secpol": "None",
        "secmode": "None",
        "login": true
    }
]

This is not the real flow I plan to have though. What I want is to poll a list of objects periodically, process the output (using a custom node Iam developing) and send to a database. I need to save the data in a loop locally and then when the last one in the loop is polled, I sent the whole data to the DB. 

But the issue is that if reading the last object has some error and no output is coming from the opcua client node, then the whole data will be lost as the data will be sent to DB only when the output of the last object has come (last object is indicated with a flag from the initial triggering node).

mikakaraila, 2023-01-23:
Console output could help me, but try with Anonymous in connection parameters:

[image]

Abhijiht, 2023-01-23:
Hi,
My issue is not connectivity. Iam able to connect to an OPC UA server. But when it is lost due to some reason, I want to get an output still. You can reproduce the case by running the sample flow I shared. Without the server at the configured endpoint, the client node will not work. In this case, the expected output is to get an output from the client node (empty payload or with an error message; preferably maintaining the fields in the input from the triggering node) for every input from the triggering node.

Abhijiht, 2023-01-24:
Hi,
Is there any way you could add an option to get an output in case of error?

mikakaraila, 2023-01-24:
This is correct way to get error, use catch node:

[image]

Abhijiht, 2023-01-24:
Hi,
I have tried that. But there is no error response from the client node either.
[image]
o error response from the client node either.

mikakaraila, 2023-01-24:
I didn´t remember that this one was not fixed ( I just fixed it):
https://github.com/mikakaraila/node-red-contrib-opcua/issues/477

I fixed it and this way you can use catch node and add needed msg property you want to set:
[image]

mikakaraila, 2023-01-24:
Please test with v0.2.295

Abhijiht, 2023-01-25:
Hi,
I tried with the latest version. But error is sent only when it tries to reconnect. Can you send an error from the client node when it receives an input and read does not happen? It would be helpful if it can maintain the fields of the input message also so that it is possible to track which message got an error, especially when there are multiple client nodes

mikakaraila, 2023-01-25:
Catch node can catch all nodes, message contains node name.
Look this one: https://nodered.org/docs/user-guide/handling-errors

Here is actual message it will generate:
[image]

You haven´t understand core problem as if client cannot connect it will not process any variable (NodeId) items.
With catch you can select your action according output.

Abhijiht, 2023-01-25:
Hi,
My understanding is that when client is not connected, it queues the inputs and only process them once it gets connected. What I'm asking is if there is any way you can make a new feature (add a checkbox to let user decide) to process the input and sent an error as output instead of queuing them.

mikakaraila, 2023-01-25:
It will duplicate code. Now node error will handle those and I am busy at work + travelling next week.

Not my first priority to implement....