mikakaraila/node-red-contrib-opcua #518: Delete stored NodeIDs out of the OPCUA Client node

offrok, 2023-01-30:
Hi, i am using a opcua client node with the action read multiple. First i inject the nodeIDS, then i externally trigger the node after a short delay of 2 seconds periodically every 1 second. If i inject the flow again, the nodeIDS will stack and it will read the same variable multiple times. How can i delete specific, or all, nodeIDs out of the Client Node? At the moment i have to restart all flows to get it working again.

mikakaraila, 2023-01-30:
You can always clear buffer (nodeIds), but do you have some console error output?

offrok, 2023-01-30:
Ok, clearitems works so that the client node will stop the reading. But when i reinject the variables as the topic to store them in the buffer and then after the 2 second delay i use msg.payload = "ALL"; and msg.topic = "readmultiple; the node doens´t output anything. When i restart all flows and do the same thing it works.

mikakaraila, 2023-01-30:
Console / debug tab could help, cannot guess what is happening...

offrok, 2023-01-31:
`[{"id":"b51fe21396af6892","type":"OpcUa-Client","z":"f57e9dbff8c530f0","endpoint":"470c6f9cf3e93289","action":"readmultiple","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","folderName4PKI":"","name":"","x":640,"y":240,"wires":[["e6ca5ac62d41264a"]]},{"id":"bf97f6572170dfe7","type":"function","z":"f57e9dbff8c530f0","name":"readmultiple","func":"msg.payload = \"ALL\";\nmsg.topic = \"readmultiple\"\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":350,"y":240,"wires":[["b51fe21396af6892"]]},{"id":"c2c94f49bb7609b5","type":"inject","z":"f57e9dbff8c530f0","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":140,"y":240,"wires":[["bf97f6572170dfe7"]]},{"id":"b8d3e0200af8cdc8","type":"inject","z":"f57e9dbff8c530f0","name":"delete sub","props":[{"p":"action","v":"clearitems","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":280,"y":320,"wires":[["b51fe21396af6892"]]},{"id":"bb3932639ec33f28","type":"function","z":"f57e9dbff8c530f0","name":"create target array","func":"\nconst opcuaConfig = {\n    \"ns=3;s=\\\"DB_OPCUA\\\".\\\"Roboter2\\\".\\\"Ampel\\\".\\\"rot\\\"\": {\n        \"label\":\n            \"Total CPU Usage\",\n        \"topicStructure\": {\n            \"group\": \"Roboter2\",\n            \"spec\": \"Ampel\",\n            \"param\": \"rot\"\n        },\n        \"settings\": {\n            \"unit\": \"\",\n            \"dataType\": 1,\n            \"description\": \"\"\n        }\n    },\n    \"ns=3;s=\\\"DB_OPCUA\\\".\\\"BinPicking\\\".\\\"Ampel\\\".\\\"gelb\\\"\": {\n        \"label\": \"Motor temperature axis 1\",\n        \"topicStructure\": {\n            \"group\": \"BinPicking\",\n            \"spec\": \"Ampel\",\n            \"param\": \"gelb\"\n        },\n        \"settings\": {\n            \"unit\": \"\",\n            \"dataType\": 1,\n            \"description\": \"\"\n        }\n    }\n}\nconst targetArray = Object.keys(opcuaConfig);\n\nconst outputArray = targetArray.map(elem => {\n    return {\n        topic: elem.toString()\n    }\n})\n\nmsg.out = outputArray;\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":210,"y":160,"wires":[["7ed2c3b466844d7f"]]},{"id":"7ed2c3b466844d7f","type":"function","z":"f57e9dbff8c530f0","name":"serialize","func":"let outputArr = msg.out;\nmsg = [outputArr];\n\nnode.warn(msg)\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":400,"y":160,"wires":[["b51fe21396af6892"]]},{"id":"f6e8046403cfa0f4","type":"inject","z":"f57e9dbff8c530f0","name":"","props":[],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":110,"y":80,"wires":[["bb3932639ec33f28"]]},{"id":"e6ca5ac62d41264a","type":"debug","z":"f57e9dbff8c530f0","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":890,"y":240,"wires":[]},{"id":"470c6f9cf3e93289","type":"OpcUa-Endpoint","endpoint":"opc.tcp://192.168.1.110:4840","secpol":"None","secmode":"None","login":false}]`

This is a simplified Flow. When i clear the items i cant subscribe to them again unless i restart the flow

offrok, 2023-02-08:
Ok i kind of figured it out. It wasn´t working with msg.action = clearitems (then i had to refresh the flow because the client node was not working at all) but with msg.topic = clearitems it worked as i intended.

closed 2023-02-08 by offrok