mikakaraila/node-red-contrib-opcua #519: syslog and deamon.log entries from node opc ua

wjomo, 2023-02-08:
Hi Mika,
thank you for your very powerful opcua nodes. It works fine.

I have got a question about two log entries from the opc-ua nodes which i could not explain to me.

I don't know where these come from and what the entries mean. I hope you can help me here. The nodes all work without error.

In my flow, an OPC UA server node and two OPC UA client nodes are used. The endpoints of the client nodes are defined via a message in the flow. One client node is currently not parameterized because it is not used. The clients work in READMULTIPLE mode.

I think the first entry "maxAllowedSessionNumber" comes from the OPC UA server node. I think I configured something wrong there. Unfortunately I don't know what I did wrong.

In the attached .txt file you will see the entries from syslog and deamon.log

Thank you very much for your help

[logentries_OPCUA_syslog_deamon_log.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/10687178/logentries_OPCUA_syslog_deamon_log.txt)

mikakaraila, 2023-02-09:
It is just warning, Etienne implemented v105 specification support into the node-opcua. Future version should use maxSessions instead of maxAllowedSessionNumber property.

[NODE-OPCUA-W21] maxAllowedSessionNumber property is now deprecated , please use serverCapabilities.maxSessions instead

wjomo, 2023-02-09:
Hi Mika,
thank you for your very quick answer. 
Now I understand thant warning.

I have one more question, where does this message come from:

> Aug 19 11:28:22 iot2050-debian-wi2 node-red[266]: 11:28:22.826Z :register_server_manager       :429   contacting discovery server backoff  opc.tcp://iot2050-debian-wi2:4840  attempt # 14  retrying in  50  seconds

This log entry occurs very often in the log files. I still don't understand where this entry came from?

Thanks a lot for your help.

Best regards

mikakaraila, 2023-02-10:
Server is trying to register itself to local discovery (LDS), turn off LDS will remove messages.

I should perhaps add discovery node to OPC UA nodes...

mikakaraila, 2023-02-10:
Seems to work, I will make new version during weekend:
[image]

mikakaraila, 2023-02-10:
Added some error handling and status is shown under node:
[image]

[image]

wjomo, 2023-02-10:
Hi Mika,
That sounds and looks good. I'm looking forward to the new version. You are doing a great job.  :+1:

Best regards

mikakaraila, 2023-02-12:
New version available. Please test & close.

wjomo, 2023-02-12:
Hi Mika, 

thanks for publishing your new version.

I tested the discovery node. First I started an OPC ua server, it reported to the Discovery Server and I was able to find it via UAExpert. Then I added a second OPC UA server (with a different port configuration) to my flow and tried to reach this via UA Expert, after which the first server that was registered could no longer be found. It looks as if only the server started last is logging on to the Discovery Server.
I think it should be possible to register more than one OPC UA server on the Discovery Server. What do you think? Maybe I misunderstood something or tested wrong?

Thanks for your help and best regards

mikakaraila, 2023-02-12:
Yes, you are correct I put wrong object into the return msg.
I fixed this:
[image]

mikakaraila, 2023-02-12:
Please re-test with new version.

wjomo, 2023-02-12:
Hi Mika,
[image]
I have now tested the new version. In my flow there are two OPC server nodes Server 1 with port 53890 the second server has port 53880.
Only one server appears in the debug output of the discovery node (port 53880). Should both be visible here?
With UAExpert I can only reach one server, also the server with port 53880.
Can it be that I configured something wrong?

Thanks for the help.
Best regards

mikakaraila, 2023-02-13:
For me different servers appear correctly. I expect it can be that the "same" server can register itself only once.
I teste with three different servers:
1) Prosys UA Simulation server
2) Node-Red-OPCUA-Server and 
3) Our server 

All three were registered correctly.

I will take a look how node-red OPC UA server will register itself. I expect it needs something...

mikakaraila, 2023-02-13:
Hmm, discovery uses applicationUrI as key so this is root cause for the fact that server with the same applicationUri is visible only once.

Could be caused by OPC Specification or node-opcua discovery implementation... any comments @erossignon ?

[image]

mikakaraila, 2023-02-20:
ApplicationUri must be unique on every server, so same server with different port is not possible to register to discovery server.

wjomo, 2023-02-20:
Hello Mika,
Thanks for your answer. If I understand correctly, only one server can be registered, otherwise the ApplicationUri is always the same if the server is running on the same hardware?

Thanks.

Kind regards

mikakaraila, 2023-02-20:
Yes, only one with the same ApplicationUri. 

Of course we could generate new ApplicationUri to test this like Server1, Server2 etc. but in normal case different servers have own always own ApplicationUri.

closed 2023-02-20 by mikakaraila